gnea/grbl #401: $ option to change min PWM value

rjkorn, 2018-03-21:
Could you add an option like $33 to set the min PWM value?

I think I fine tuned my laser setup pretty well but certain materials need a different value.

Not a major issue. GRBL been great for years........

tjhub3, 2018-03-21:
There is a minimum duty cylce component to the spindle PWM generator, no? Controlled by $31.

For example: If your laser can't operate with PWM values <5% you should set $31=50 (assuming you're using the default 0-1000 speed&PWM scale).

This will prevent grbl from gating/executing a PWM signal if it below a 5% value. Is that what you want?

Otherwise you can accomplish this in your CAD/GCODE generation program. I know LightBurn, for example, has a minimum layer specific PWM component to prevent rastered images from falling below a certain power level.

Are you looking for something that isn't covered by either of those solutions?

rjkorn, 2018-03-22:
Thanks for the reply.

I thought the $31 value behaves the same for $32=0 or $32=1 usage. Is that not so?

Does setting $32=1 change the implementation of $31?

The min pwm I set in config.h...

$31 is good for calibrating spindle speed to lowest spindle PWM output
$31 is min speed value to allow

Heres how I set my dc spindle speeds 

$30 = 10,000 and $31=1000

Then 
M3S0 = PWM 0V
M3S10 = PWM .02V
M3S500=PWM .02V
M3S1000=PWM 0.2V
it starts moving in a linear fashion from there up to  
M3S10000=PWM 5V 
M3S24999= PWM 5V

By setting min spindle pwm=128 in config . h I now get
M3S0 = PWM 0V
M3S10 = PWM 2.5V
M3S128=PWM 2.5V
M3S1000=PWM 2.5V
it starts moving in a linear fashion from there up to  
M3S10000=PWM 5V 
M3S24999= PWM 5V

Its the bottom end that I need raised.

You lose half your resolution with this example.

I can us op amp to translate levels to what I need with no  loss of resolution but I was looking for a simpler firmware solution because I can lice with as little as 100 discrete steps..

I find it easy on spindles since and have just started playing with GRBL and lasers. 

For diodes I need the same level shifting. For CO2 I prefer the op amp method. lets me have manual gain controls to tweak....

I have plenty of issues with lightburn but I use a postprocessor script that fixes them all. 

I use Vectric Aspire for real work but at home its Inkscape/LightBurn 

Things like controls M7 M8 for Exhaust fan and Laser coolant Water Pump, feed rate tweaks M3 control when needed.....

I hope pre and post script support will be added to LB soon but thats seems doubtful since the author only seems to want to support some M codes in there but no G codes. But I've posted that in their forum...

chamnit, 2018-03-23:
Don't worry. This is already been planned to be configurable in the next version of Grbl.

closed 2018-03-23 by chamnit