gnea/grbl #402: How is TLO supposed to work in GRBL?

bostonbladeco, 2018-03-22:
I have a z setter probe in a fixed location in machine space connected to the probe pin and despite my best efforts I cannot seem to get it to set the tool length offset TLO. I have reviewed the available bug reports and documentation.  I am using the most recent version of GRBL. I am not sure if I am hitting an issue or if I am not understanding how TLO is supposed to be set and work.

The most frequent problem is that I run into "unsupported command" errors when issuing a G43.1  command such as G43.1 Z0.250 . That only seems to be accepted without error if immediately preceded by something like "T1 M6 G43" or anything else with M6. Checking $# after the TLO variable shows no change.

Is there a simple gcode example that works with GRBL for TLO setting, is this an issue that I should turn into a formal bug report or what else am I missing?

Thank you.

jahnj0584, 2018-03-23:
change the macro to G5_ G38.2 G10 P0 L20 Z__ 
Where the Z value is the difference from the probe to the table top. G5_ is a work area you save for the probes location.

bostonbladeco, 2018-03-23:
Thanks for the prompt feedback. A couple follow ups...

If T1 M6 G43 is not supported why doesn't it generate an error when issued instead of returning ok?

Why does G43.1 sometimes return "unsupported command" and sometimes not? What are the preconditions for using it?

So then it would be a bug or issue if G43.1 Z0.250 were issued and there was no change to the TLO value in $# ? I will make a detailed separate report in such a case

zeevy, 2018-03-23:
For me G43.1 Z(offset_value) never gave me "unsupported command" error.
And M6 or G43 always gives "unsupported command" error
$# always has the applied TLO value.

Which GUI are you using?

chamnit, 2018-03-23:
FYI, Grbl has never supported G43 or M6. It only supports G43.1. Your GUI will sometimes interpret G43 or M6 for you and convert it to something that Grbl understands. If you are still having problems, connect to Grbl directly through a serial terminal and show us the exact set of commands you use to make the error. There should be no problems.

bostonbladeco, 2018-03-23:
I was on serial console, I will attempt to produce a complete bug report with the relevant details but am away from the machine at the moment. Thanks.

closed 2018-03-23 by bostonbladeco