gnea/grbl #407: Noob with questions on how to contribute

DaveSkowron, 2018-04-01:
Hi. I'm new here and have a question. I want to contribute a file for people to use if they want. I've gotten a lot from this place and want to give back a little. It is a Corel version of the PCB for the limit switches. People who wish to laser etch using the paint method can use it. It's got the vector outlines of all the traces and pads plus the rastered area around them. How do I provide the file? Or is it even allowed, or for that matter, would anyone even care to have it?

Thanks in advance.

DaveSkowron, 2018-04-01:
Here's the file:

[grbl limit switch pcb v1.pdf](https://github.com/gnea/grbl/files/1866760/grbl.limit.switch.pcb.v1.pdf)

closed 2018-04-02 by DaveSkowron

X3msnake, 2018-04-04:
What did you use to design the board?
Can you share the design file?

X3msnake, 2018-04-04:
> Corel 2017. I couldn't upload the design file but you should be able to
> deconstruct the pdf.

Oh ok, thought you had used a IDE of some kind :)