gnea/grbl #409: homing feedback

J-Dunn, 2018-04-08:
Feature request.

It would be useful in assessing the repeatability of homing sensors if the homing command return a resume of the changes from the last homing position ( or the turn-on state ). 

This could be returned as a comment to avoid confusing any listening software.

chamnit, 2018-04-08:
Yes. This has been requested long ago and is in my development notes. My current thought has been being able to configure your limit switches as a probe input. I may also add an optizonal difference statement after each homing cycle to report what you want.

closed 2018-04-08 by chamnit

J-Dunn, 2018-04-09:
If you are referring to dual X drive , GRBL only supports three axes. 

If you have two identical X drives running from the same step signals there is no way to independently drive one side to 'square up' .  Maybe you need to explain how you see this working.