gnea/grbl #416: GBRLUpload issues

Dhammika1986, 2018-04-16:
Hi Friends,

I'm doing DIYs as a hobby and has some knowledge on Electronics, C programming and PIC Micro-Controllers. I used Arduino Uno in my first CNC project and when I wanted to upload grblUpload program to the arduino, I faced few issues with gbrl directory. To resolve that I removed the gbrl folder from 'Documents\Arduino\libraries' and instead placed in 'C:\Program Files (x86)\Arduino\libraries'. Then the program loading was success with below massage;

_Sketch uses 30562 bytes (94%) of program storage space. Maximum is 32256 bytes.
Global variables use 1633 bytes (79%) of dynamic memory, leaving 415 bytes for local variables. Maximum is 2048 bytes.
Low memory available, stability problems may occur._

Then I loaded G-Code Sender. While the baudrate is 115200 and Uno connection are ok, settings for GRBL window was blank at all. When I tried work with 'Machine Control', it worked only for 9 commands only. When tried to send a file, it sent only 9~11 lines only and stucked. 

Highly appreciate if any one can help me out to fix the issues..

Dhammika1986, 2018-04-16:
When I had Grbl directory in 'Documents\Arduino\libraries\' , below is the error I received in trying to upload the program in Arduino,

_C:\Users\E  NET\Documents\Arduino\libraries\grbl\examples\grblUpload\grblUpload.ino:27:18: fatal error: grbl.h: No such file or directory

 #include <grbl.h>

                  ^

compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Uno._

chamnit, 2018-04-16:
It looks like you flashed Grbl fine. You just need to use a GUI that supports Grbl v1.1. You can also verify that Grbl works through the Arduino terminal.

I'm not exactly sure why your compilation failed when you placed your library in your user library and worked fine when you placed it in the program library directory. Arduino has been switching up its internal structure a bit, but it more likely has to do with something like a typo.

closed 2018-04-16 by chamnit

Dhammika1986, 2018-04-16:
Thanks Chamnit. I used grbl 1.1 but the GCodeSender is 1.0,7
After Your comment, I used GCodeSender is 1.0.9, that works fine, and I can go ahead. Thank you very much for the support.