gnea/grbl #417: Move axes independent from each other

mikemaikaefer, 2018-04-17:
I'm looking for a possibility where all axes move independent from each other and therefore also may stop at different times. So when a motion command is given, they do not "draw" a line from current to target coordinate position, but move to their target positions each at its maximum speed.

In my opinion command G0 should do that, but here the motion speed of axes may be limited in order to let all of them arrive at the same time (so this would be a bug). So there is no real difference between G1 and G0, the last one should let move each axis as fast as possible (which isn't the case), but I also would be happy with a special mode which enables this feature.

neilferreri, 2018-04-17:
Just send your commands for each axis on different lines.
Gcode has been this way for a long time.  Grbl follows a well proven standard.

Sortland, 2018-04-18:
@neilferreri then the axec move sequentially and no longer parallel

@chamnit which version supports this correct G0 command? I understand your argumentation bit do not agree with this, when G0 works in this way there is no difference to G1, one of both would be useless.

chamnit, 2018-04-18:
@Sortland : I disagree. G0 is different from G1. G0 always runs at a max feed rate, where as G1 follows the programmed F feed. This allows you to switch between the G0 max speed vs G1 cutting speed quickly, especially during manual control. How G0 is implemented is really up to the manufacturer. The behavior of G0 isn't consistent across all CNC machines and maybe be configurable in some cases. Go check for yourself. Again, Grbl isn't changing this anytime soon and will only do it in a "pro" version far in the future.

There are other firmwares that allow you to program a separate F for G0. That is completely wrong and would make G1 and G0 pretty much the same.

109JB, 2018-04-19:
Actually, on industrial machines that use the dog-leg motion for G0 there may or may not be any difference in the time it takes to reach the end position when compared to a linear move.  It depends on how the machine "thinks".  Lets say you have to move 1000 mm in X and 2000 mm in Y and both X and Y move at 1000 mm/min.  If the machine places no further limits on either the dog leg motion of the linear motion, then the machine still has to move the Y axis 2000 mm, which will take 2 minutes regardless of the path. 

The only time when there would be a difference is when a particular controller has a limit for linear interpolated moves. For example, lets say the machine above limits linear interpolated moves to a 1000 mm/min overall rate.  In this case, the linear G0 will take longer because the Y axis can no longer move at its max rate of 1000 mm/min.  The dog-leg move in these controllers is not really considered a linear interpolated move, so that 45 degree initial move of X and Y simultaneously can occur at an overall rate higher than 1000 mm/min

GRBL doesn't limit this way and will move as fast as possible to the destination. For example, with 1000 mm/min maximum feed rates on both X and Y, a 45 degree move (ie: x 500 and y 500 simultaneously) results in a overall rapid feed rate of 1414 mm/min, which is both axes moving at their max rates of 1000 mm/min (sqrt(1000^2 + 1000^2) = 1414 mm/min). So, even though Grbl is making a linear interpolated move, it is still moving each axis at 1000 mm/min, so in this case with each axis having the same maximum speed, the limiting factor will be the axis that has to move the farthest, but dog-leg or not the time to reach the end point will be the same.

If you want to further complicate things, there are machines now that have an option to use an arc motion for rapid traverse (G0). This results in gentler acceleration and deceleration and is easier on the mechanicals of the machine.

chamnit, 2018-04-19:
@109JB: You're right. There should be no to little time difference in time between moves. A move, either dog-leg or linear, is still limited by the slowest axis. I think there are some instances that either one could be slightly faster than the other. For example, if one axis has a limited acceleration while another is limited in speed, a dog leg could be faster.

I also talked with our machinist at my workplace after posting my last comment to verify a few things. He stated the opposite of what I recalled. Most higher end CNC machines move now with a linear motion for G0. Most cheaper production CNCs and older hobby mills do the dog-leg. And that there is no configurability between the two types of motion that he has ever seen. Usually a manufacturer picks one type.

109JB, 2018-04-19:
FYI, 

1.  G0 is defined as a rapid traverse move, and G1 is defined as a linear move at a defined feed rate.  There are many industrial machines that have a slower limiting speed on "feed" moves compared to rapid moves.  This is true regardless of dog-leg patterns, arc patterns, or linear patterns.  For example one machine may have a 1000 in/min maximum rapid rate but a 300 in/min maximum feed rate. So a G0 programmed move could go at 1000 in/min, but the same move with G1 could only get 300 in/min.

2. Many industrial machines ONLY use a linear interpolated G0 move.  So saying that a controller that makes a machine move in a linear interpolated fashion during G0 is doing it wrong is saying that those machines in industry are wrong as well. This isn't the case and as a matter of fact, if you look at the RS274NGC standard, G0 is defined as a "Rapid Linear Motion". So, this is one place where manufacturers have deviated from the RS274NGC standard by implementing the dog-legs and arc rapids.

3. In my view a G0 move should be linear as it is predictable what it will do, however, it can be argued that it should simply get to the programmed position as fast as possible.  In the case of Grbl, it gets there just as fast using a linear move as it would a dog-leg move. So, since there is no time difference, I see absolutely no benefit from using a dog-leg move compared to a linear move.

109JB, 2018-04-19:
@chamnit The Fanuc controller are the ones I am aware of that have a parameter that can be set to choose the rapid method as either a linear move, or a dog-leg.  Even the old Cincinnati VMC's I used to run used a linear interpolated rapid.  The newer Brother VMC's are the ones that I have heard about the arc moves for rapids to reduce machine wear, and there again is a parameter that can be set.