gnea/grbl #418: z-axis not responding during homing

karlbrown90, 2018-04-18:
I have definitely missed a setting somewhere .
I have at last thrown in the towel and need to admit I cannot find an answer on my own.
I am hoping someone will be kind enough to point this non-programmer in the right direction.
The problem is that when I initiate homing ($H) the spindle moves possibly 2mm in the positive direction ( the correct direction for my application), performs a pull-off motion and then stops. This happens no matter its starting position, meaning it never actuates the z limit switch. 
During the debugging I did try adjusting the travel but it had no effect.
I am running Grbl 1.1 on an Arduino and sending using bCNC.
I have installed limit switches to my cnc build. two switches at the end of each axis ( i.e. Zs, Zn, Ys,Yn and Xs,Xn).  I commented out line 339 ( VARIABLE_SPINDLE) in order to enable d11 and have access to my z-limit switch. I have verified that bCNC sees all of the switches.
I am currently running with default settings, except for the disabled config.h setting. 
any suggestions would be greatly appreciated.
thx

jahnj0584, 2018-04-18:
Invert the pins

karlbrown90, 2018-04-19:
Invert pins? 
D11 and D12?
commenting out VARIABLE_SPINDLE swaps D11 and D12. 
I verified that the arduino recognizes z switch.

jahnj0584, 2018-04-19:
In the software. I think it's $5

karlbrown90, 2018-04-20:
Thanks jahnj. tried but it didn't work.
I have no real idea what is going on but I have found an acceptable "work around".
In my project the negative end-stop pins do not want to be hard limits. I simply removed switches from the negative pins and setup soft stops for negative extremes. 
When I have the time I will try to re address this issue. will post what I find.

chamnit, 2018-04-25:
You have electrical noise in your limit switch lines. Try operating the switches manually without any power to anything but the Arduino. You should see it work fine. If so, you'll need to address the noise issue.

closed 2018-04-25 by chamnit