gnea/grbl #422: External jog controller

Jibeji, 2018-04-25:
Hello,
I am currently building an autonomous jog controller. The pendant uses an ATmega328p which reads a quadrature encoder and some switches, it will deliver $J Gcode instructions.

I am wondering what is the best way to send those instructions to GRBL.
Should I switch TX and RX lines while using the pendant or is there a better option?

swarfer, 2018-04-25:
send them to the GUI, then the GUI passes them along to GRBL

terjeio, 2018-04-25:
Similar to this?

[Grbl Dro/Mpg](https://youtu.be/_0Nmh164dvo)

I am "listening in" to grbls real time reports in the second processor. To change to MPG-mode I have connected a switch to the grbl processor (will be momentray in final version). If grbl can switch to MPG mode (is in IDLE state) it adds MPG:1 to the real time report telling the second processor to take over the input stream to grbl. This can be done by a switch, however my implementation uses a secondary UART input. When switching back to normal mode I add MPG:0 to the real time report.

terjeio, 2018-04-25:
The second UART stream is input only, the primary always handles the output stream so that is connected in parallell to the PC and the MPG processor.

Since I am using function pointers for grbl access to the HAL (in my ARM port) I switch the input stream to the secondary UART by just swapping the relevant pointers, here is how I do it for TIs MSP432 processor for now:

``` c
static void modeSelect (bool mpg_mode)
{
    BITBAND_PERI(MODE_PORT->IE, MODE_SWITCH_PIN) = 0;
    BITBAND_PERI(MODE_PORT->IES, MODE_SWITCH_PIN) = !mpg_mode;
    BITBAND_PERI(MODE_PORT->IFG, MODE_SWITCH_PIN) = 0;
    BITBAND_PERI(MODE_PORT->IE, MODE_SWITCH_PIN) = 1;

    // Deny entering MPG mode if busy
    if(mpg_mode == sys.mpg_mode || (mpg_mode && (gc_state.file_run || sys.state != STATE_IDLE)))
        return;

    BITBAND_PERI(MODE_PORT->OUT, MODE_LED_PIN) = mpg_mode;

    serialSelect(mpg_mode);

    if(mpg_mode) {
        serial2RxFlush();
        hal.serial_read = serial2GetC;
        hal.serial_get_rx_buffer_available = serial2RxFree;
        hal.serial_reset_read_buffer = serial2RxFlush;
        hal.serial_cancel_read_buffer = serial2RxCancel;
    } else {
        serialRxFlush();
        hal.serial_read = serialGetC;
        hal.serial_get_rx_buffer_available = serialRxFree;
        hal.serial_reset_read_buffer = serialRxFlush;
        hal.serial_cancel_read_buffer = serialRxCancel;
    }

    // Report WCO on first status report request from MPG processor
    if(mpg_mode)
        sys.report_wco_counter = 1;

    // Force a realtime status report
    hal.protocol_process_realtime('?');

    sys.mpg_mode = mpg_mode;
}
```
IMO the same can be achieved by using a physical switch on the UART RX signal.

Jibeji, 2018-04-26:
When will it be released ?
Just to know if I continue my own project in the meantime.

terjeio, 2018-04-27:
Have a look [here](https://github.com/shooter64738/grblRTC/issues/2?_pjax=%23js-repo-pjax-container#issuecomment-384992981) and join in if you are interested...