gnea/grbl #423: Jog cancellation (0x85) sometimes not working (again)

zeevy, 2018-04-25:
#95 

I ran into this problem even though i am not sending the jog commands after jog cancel (0x85)
Check the video.
If i press and hold the jog button gui will send continues jog commands one after another (sends next command only after receiving the ok)
This way it fills up the planner buffer, then i hit the jog cancel button.
Then planner buffer shows 14, then grbl continues to jog. I also checked in the gui log to make sure no jog command is sent after jog cancel.

https://www.youtube.com/watch?v=-S8nRhSO57A
[image]

zeevy, 2018-04-25:
#95

zeevy, 2018-04-25:
Well this is not exactly duplicate of the issue 95.
The proposed solution will not work here. 
Here I am not sending any new jog commands after jog cancel, but still motion is not stopped grbl continues to execute the last command in the planner buffer

chamnit, 2018-04-25:
The problem is caused by the same root issue.