gnea/grbl #426: How to connect TTL Laser Module with arduino uno

MuhammadFaisalEhsan, 2018-04-29:
I am  using a  laser module which I bought from aliexpress .Link regarding that laser is given below:
 
( https://www.aliexpress.com/item/Laseraxe-450nm-5500mW-5-5W-DIY-Desktop-Mini-Laser-Engraver-Engraving-Machine-Laser-Cutter-Etcher-35X50cm/32570563136.html)

 I am not using  any cnc shield in my CNC Laser cutting machine.I have used easydrivers for driving my sepper motors but i have problem in connecting that laser module.  Can i make direct connections of that laser module to my arduino uno( which i have made grbl controller) .If yes then at what pins?

JBSchueler, 2018-04-29:
Please don't post the question multiply times....
You already asked this question in [this](https://github.com/gnea/grbl/issues/257#issuecomment-385231579) topic...

MuhammadFaisalEhsan, 2018-04-29:
When i asked question in that topic you mentioned  out i didn't provide the proper specifications of laser module i m using . So I raised an issue by giving  details in link.Dear Ben Schueler would you now help me considering the scenerio i provided in this issue.

MuhammadFaisalEhsan, 2018-04-29:
Thanks Ben Schueler  but i still have a doubt which is 
 for laser i am using  whether a TTL driver is must or not?I t is DIY Laser engraver .You can see link i have given while raising this issue.Would you please confirm it to me?

closed 2018-04-29 by chamnit

sz3bbyla, 2019-01-10:
My 3d printer is on arduino due (RADDS board) and pinout are on 3 V
how to connect this ?