gnea/grbl #429: Enable I2C for a Spindle control

SailWithChips, 2018-05-02:
I think it was issued in old GRBL (0.9) but not "solved" at all. I would like to control the rpm of the spindle via i2c DAC since the frequency inverter is sensitive to the PWM frecuency (working with atmega328p) and the setpoint for the rpm's are not so stable. It could be really nice to have the chance to output via i2c the rpm setpoint.

Harvie, 2018-05-27:
Just add low-pass RC filter to your PWM to filter it?

chamnit, 2018-05-27:
Adding i2c spindle control was not “solved” because it was never on the development path. This is certainly an idea to add eventuallly but there are plenty of other things that are much higher priority.

closed 2018-05-27 by chamnit

SailWithChips, 2018-05-27:
@Harvie already have a low pass filter but is still not a fixed signal, and as you know it is not as fast as an i2c would be since a low pass filter put a delay into acceleration to setpoint rpm’s.

@chamnit, thanks to answer anyway, hope to have this feature in the future... i can imagine there are some features with higher priority.

Harvie, 2018-05-27:
why would you need to change spindle speed so rapidly? i think the inertia of spindle is limiting factor anyway...