gnea/grbl #436: How to change PWM Frequency.

StuartB4, 2018-05-11:
Just wondered if anyone knows which part of the **cpu_map.h** file I need to edit to change the PWM
frequency from 1khz to 8khz. I found references to it but can't find anywhere where it actually tells you what to edit.
Just not sure what to do.

My thought was to comment out line 139 with //  and remove the //'s from line 141.
Any help would be greatly appreciated.

138  // Prescaled, 8-bit Fast PWM mode.
139  **#define SPINDLE_TCCRA_INIT_MASK   ((1<<WGM20) | (1<<WGM21))  // Configures fast PWM mode.**
 140 // #define SPINDLE_TCCRB_INIT_MASK   (1<<CS20)               // Disable prescaler -> 62.5kHz
141  **// #define SPINDLE_TCCRB_INIT_MASK   (1<<CS21)               // 1/8 prescaler -> 7.8kHz (Used in v0.9)**
142  // #define SPINDLE_TCCRB_INIT_MASK   ((1<<CS21) | (1<<CS20)) // 1/32 prescaler -> 1.96kHz
143  #define SPINDLE_TCCRB_INIT_MASK      (1<<CS22)               // 1/64 prescaler -> 0.98kHz (J-tech laser)

chamnit, 2018-05-18:
@StuartB4 : In the future, please don't post the same issue in multiple places. This is a clone of https://github.com/grbl/grbl/issues/1398#issuecomment-388408024

closed 2018-05-18 by chamnit

StuartB4, 2018-05-18:
I did'nt understand why there was 2 different places. Sorry.