gnea/grbl #437: ALARM:1 if triggering a solenoid valve via saintsmart relay board

momo-aux, 2018-05-12:
Hello,
I'm using a Arduino nano with Grbl 1.1f connected to a OrangePi Zero via usb.

I created a little test script in python that sends simple GRBL commands over serial to move a Cocktail shaker in x +- direction.

All works fine and the relay switches (using relay board connected to the GPIO Pins on the OrangePi) work.
I have a homing switch connected to D9 on the nano with a external 1kR Pullup 
No capacitors so far.

But if I connect the Solenoid valves (12v 500mA) to the relays. As soon as one opens I get ALARM:1 on the serial.

I tried to change wires, change how I power the solenoids and the relay board. No changes.
I can't get it to work if the valve really opens. Without the valve (blank relay) everything works.

This is the relay board: https://www.sainsmart.com/products/8-channel-5v-relay-module

Can somebody give me a hint how to fix this?

`
[VER:1.1f.20170801:]
[OPT:V,15,128]

$0=10
$1=25
$2=0
$3=1
$4=0
$5=0
$6=0
$10=0
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=1
$24=25.000
$25=1500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=40.200
$101=250.000
$102=250.000
$110=5000.000
$111=500.000
$112=500.000
$120=250.000
$121=10.000
$122=10.000
$130=600.000
$131=200.000
$132=200.000
`

PlifPlafPlouf, 2018-05-13:
Hi,
Place a signal diode on the solenoid poles (cathode on the +, anode on the -), and also small cap (around 10 nF)

momo-aux, 2018-05-13:
OMG, you are right.
The diodes on the relay board are no flyback diodes... have to add my own...
Thank you for that :-)

closed 2018-05-13 by momo-aux