gnea/grbl #438: Recompile of GRBL for use on RPI3 with a Protoneer CNC Hat

yisabell, 2018-05-15:
I apologize for my confusion, but I need to recompile GRBL due to a small change I require in the config,h file (need to remove the Y axis for use with a laser)

I see most of the document refer the Arduino and I do have the Arduino IDE installed on my Windows machine, however I am not sure of what, where or how to move the compiled code to the RPI3.

Any help would be appreciated.

langwadt, 2018-05-15:
I would guess https://wiki.protoneer.co.nz/Raspberry_Pi_CNC#Software_.28Firmware.29

yisabell, 2018-05-15:
> 28Firmware.29

I am going blind from reading it over and over.

The challenge is that I need to change the config.h, then compile, them
flash it onto the hat.

I think I need more details on where GBRL source code and the process to
compile and flash.

Thanks in advance for all the help.

Yves Isabelle
905-407-1071
131-2125 Itabashi Way
Burlington Ontario
L7M 0A1

Protoneer, 2018-05-15:
@yisabell , the RPi image that we have only uploads pre compiled hex files. 

Arduino is pre-installed and you will have to follow the GRBL compile steps if you need to build a custom hex file.

[https://github.com/gnea/grbl/wiki/Compiling-Grbl](https://github.com/gnea/grbl/wiki/Compiling-Grbl)

The Firmware folder on the desktop has thas example script of how you can upload the hex file or you can upload it directly from Arduino.

yisabell, 2018-05-15:
Thanks Bertus,

The piece I was missing/unaware was that an ATMEL is built in to the hat
that makes it essentially and Arduino.  Also that the Arduino IDE was
already built into your most current image v4.10.

I followed the compile process defined in  https://github.com/gnea/grbl/
wiki/Compiling-Grbl, the first time it gave me some piping errors so I ran
it again and it completed properly.

Mow the homing process started, but I have to look at one of the axis which
just rattle after hitting the switch. It may be electrical or noise.

But thank you all for helping me get passed this hurtle.

Yves...

closed 2018-05-18 by chamnit

yisabell, 2018-05-19:
Thanks for the support.