gnea/grbl #440: Global interrupt state assumed in eeprom_put_char function

scottnortman, 2018-05-17:
Hi,
I noticed in the eeprom_put_char function the global interrupts are disabled (line 79) with  a cli() call without preserving the prior state of SREG, and at the end of the function, a call to sei() is made.  This will therefore enable global interrupts, even if they were not enabled in the call to cli() on line 79.

Not sure if this matters, but just thought I'd mention it...

chamnit, 2018-05-18:
Sure about this? The eeprom_put_char() function is directly from the Atmel Framework. I believe it's a direct copy of their code.

scottnortman, 2018-05-18:
Yes, on line 79 of the file eeprom.c, the first function call in
eeprom_put_char is

cli();

this is called without a prior call to

uint8_t sreg = SREG;

as you do in other places in the code.

And the last call in eeprom_put_char.c is

sei();

instead of

SREG = sreg;

as done in other places in the code.

--Scott

chamnit, 2018-05-18:
@scottnortman : Are you saying that this function from the Atmel Software Framework has a bug in it?

scottnortman, 2018-05-18:
Yes, it appears that is the case.

cri-s, 2018-05-22:
I have placed

CRITICAL_SECTION_START
and
CRITICAL_SECTION_END

at begin and end of serial_init and settings_init functions and using avr
eeprom_update_byte and eeprom_read_byte as eeprom functions

This are defined as follow inside nuts_bolds.h

#ifndef CRITICAL_SECTION_START
#define CRITICAL_SECTION_START  unsigned char _sreg = SREG; cli();
#define CRITICAL_SECTION_END    SREG = _sreg;
#endif //CRITICAL_SECTION_START

It is required inside serial too because it access 16bit variable and
avr use a global shadow register for 16bit access, only important if
there as example i2c support.

chamnit, 2018-05-22:
@cri-s : FYI, this abstraction is already present in the unpublished re-write for ARM. I won't be making any changes to this code base unless there are bugs to fix.

closed 2018-05-22 by chamnit

scottnortman, 2018-05-22:
Hi, Understood.

I'm actually working on a port for the Cypress PSOC 6, which is a cortex m4
+ m0.