gnea/grbl #441: Sketch too large for Arduino Uno

yihan0512, 2018-05-17:
Hi there, I followed the adding zip library instructions to add the grbl library and attempted to upload the grblUpload sketch. However, I got an error says sketch is too big:

`Sketch uses 52,238 bytes (161%) of program storage space. Maximum is 32,256 bytes.
Global variables use 1,633 bytes (79%) of dynamic memory, leaving 415 bytes for local variables. Maximum is 2,048 bytes.
processing.app.debug.RunnerException: Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
	at cc.arduino.Compiler.size(Compiler.java:336)
	at cc.arduino.Compiler.build(Compiler.java:159)
	at processing.app.SketchController.build(SketchController.java:643)
	at processing.app.Editor$BuildHandler.run(Editor.java:1795)
	at java.lang.Thread.run(Thread.java:745)
Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
`
Thank you very much if someone can point out the possible steps I'm doing wrong or have encountered this same error.

chamnit, 2018-05-18:
Tested the instructions on Arduino 1.8.5 on OSX. No problems. Just some redefine warnings that can be ignored. Try a fresh install of your Arduino IDE and ensure all of the board settings are correct.

closed 2018-05-18 by chamnit

yihan0512, 2018-05-22:
Found out it's due to the compiler flag, with default -Os flag there is no problem.

Harvie, 2019-11-27:
I have the sam problem with 40eb439 v1.1h.20190825 (also tried older tags and problem still persists). Arduino IDE says this:

```
Sketch uses 35576 bytes (115%) of program storage space. Maximum is 30720 bytes.

Global variables use 1633 bytes (79%) of dynamic memory, leaving 415 bytes for local variables. Maximum is 2048 bytes.
Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
Error compiling for board Arduino Nano.
```

Where do i change gcc options?

cri-s, 2019-11-27:
What gcc version do you use ?

Harvie, 2019-11-27:
```
/usr/bin/avr-g++ --version
avr-g++ (GCC) 9.2.0
```

zygmunt, 2019-11-27:
My few cents:
In general something got change in gcc, I hit into that problem with LUFA, adding LTO plugin helps, but!
Keep in mind that LTO will wipe out eeprom.
My links:
https://github.com/abcminiuser/lufa/issues/149
https://github.com/abcminiuser/lufa/issues/154
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=91189

cri-s, 2019-11-27:
> abcminiuser/lufa#149 <https://github.com/abcminiuser/lufa/issues/149>
> abcminiuser/lufa#154 <https://github.com/abcminiuser/lufa/issues/154>

Use a older Version.
You should use 4.6.0 or higher because of C++11 features.

Arduino 1.6.9 have  version 4.8.1 as example and have flto.
Oder Version have some bugs.

BR
Chris