gnea/grbl #442: CNC Bluetooth

RobertoRmz, 2018-05-17:
Hello, Good afternoon, I write because I have a doubt, make a small cnc to draw, but I want to handle it with bluetooth. I use universal G code Sender version 1.0.9. using arduino one and cnc shield V3.0

But noce where to start because I have never used Bluetooth, I have been researching this issue and what I see is the Bluetooth module HC-05.

Does anyone already have a recommendation where to start or if someone already used Bluetooth module for a CNC?

zeevy, 2018-05-18:
For that you need to change the buad rate of the bluetooth module.
check these links
https://github.com/zeevy/grblcontroller/wiki/Bluetooth-Setup-HC-06
http://www.buildlog.net/blog/2017/10/using-the-hc-05-bluetooth-module

Universal gcode sender does not support bleutooth directly, unless it supported on the host computer. 
You can try the mobile version. 
https://github.com/zeevy/grblcontroller

closed 2018-05-18 by chamnit

RobertoRmz, 2018-05-18:
Thank you very much Zeevy, for what I'm doing is first turn on and off a led, I get it with the diagram that comes in the second link. But the Arduino program to configure the Bluetooth module uses one that comes in the examples

#include <SoftwareSerial.h>
SoftwareSerial mySerial(10, 11); // RX, TX
void setup() {
  // Open serial communications and wait for port to open:
  Serial.begin(9600);
  mySerial.begin(38400);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }
  Serial.println("Configuracion");
}
void loop() { // run over and over
  if (mySerial.available()) {
    Serial.write(mySerial.read());
  }
  if (Serial.available()) {
    mySerial.write(Serial.read());
  }
}

And the configuration that I used for the HC-05 module
AT + ROLE = 0
AT + POLAR = 1.0
AT + UART = 115200,0,0
AT + RESET

The problem that I have hour is that to load program, I have to close the arduino and return it open, and probe with the ignition of a Led and if it worked, I think that if it works with a led and would be to place it in the CNC Shield so that can work through Bluetooth the Universal Gcode Sender but will continue to make tests and thank you very much for the information, it was very helpful