gnea/grbl #443: Reduce GRBL code to add LCD on Arduino UNO

Chip1920, 2018-05-18:
Hi everyone.
I need to know what i need to do to reduce the grbl code, in order to add a lcd display. 
I managed to free pins of end course and enable to use in display, but the minimal size of the display code is 1k.
Thanks.

JBSchueler, 2018-05-18:
Code reduction will be hard in my opinion.
Is using a ATmega2560 instead of an ATmega328 an option (if you using an ATmega328)?

biasedlogic, 2018-05-18:
That fits in the category 'how to make some place in a wrist watch to fit a cuckoo in'. GRBL code is packed and tight on timing, squeezing lcd in is going to be a masterpiece work, if possible at all.
You may have more luck with some older release, at the cost of limited performance/feature set of an older GRBL revision.

JBSchueler, 2018-05-18:
Another solution could be to use an STM32

https://github.com/usbcnc/grbl

chamnit, 2018-05-18:
@Chip1920 : You need a bigger chip.

closed 2018-05-18 by chamnit

cri-s, 2018-05-18:
Basically you need a ms timer, timer2 in this case.using 64 prescaler.
for 1ms interrupt or 1024 for 8ms interrupt. The timer counts in 0.5ms
or 4ms increments,
as timer0 clear timer2 and increments the tick counter if bit7 of timer2 is set
and timer2 interrupt increments the tick counter  two times. This for
saving interrupts
during faster moves.
Basically this gives 64ms on 8bit variable. The code inside
protocol_execute_runtime checks bit8 of the tick counter and if set,
it increments
ticks and reset tick variable.This gives 30.5 tick per seconds and if
divided by 6
it gives  5 hz updates.
The code for lcd is small, what do you do if it is 1kb, do you use Arduino code,
that relay on ms and timer0 not working inside grbl ?

Chip1920, 2018-05-18:
I am using the display driver from TeaCup 3D Printer. It works sending byte to byte every 10mSec in main loop, or any time i choose. I wiil try with gbl 0.9, because this code have almost 2k free, and i will increase this reducing report.c, like in grbl 1.1 version.
Anyone know where I download the old source versions of the GRBL? I only find the firmware.

chamnit, 2018-05-20:
I don't recommend trying to hack away parts of Grbl, especially an older version, to try to get an LCD crammed onto the same chip. You are just going to be in for a world of hurt and a waste a huge amount of time. Grbl is highly optimized to run well on and fit into the 328p AVR chip. Grbl also doesn't use any Arduino code and won't work with any Arduino libraries. It's much easier and more reliable to just write a host with LCD support on a separate controller. You can use any IDE for that.