gnea/grbl #444: M30 doesn't stop the PWM output

Jibeji, 2018-05-20:
Hi,
M30 command doesn't stop the PWM output, is it a feature or a misconfiguration my side?

chamnit, 2018-05-20:
Works for me. With laser mode on and off. Always try to eliminate the UI as the source of the problem by testing a direct serial connection to Grbl. Sometimes UI scrub certain gcodes and insert others. 

If you still have problems, post exact details of how I can reproduce the problem on my end with a clean install of Grbl on an Arduino only.

Jibeji, 2018-05-20:
Please find the Gcode attached.
I confirm M30 doesn't stop the PWM output on my unmodified grbl

chamnit, 2018-05-20:
Nothing attached and I need more than that to reproduce the issue. Settings, build info, a list of anything else you have changed, and how you are streaming .

Jibeji, 2018-05-21:
Hello,
There is no specific configuration for GRBL, I have downloaded the ZIP and
loaded it on a UNO board without any modification.

Here are the settings:

Grbl 1.1f ['$' for help]
>>> $$
>>> $G
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 1    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 3    (Status report options, mask)
$11 = 0.020    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 3    (Homing direction invert, mask)
$24 = 100.000    (Homing locate feed rate, mm/min)
$25 = 1500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 2.000    (Homing switch pull-off distance, millimeters)
$30 = 5000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 320.000    (X-axis travel resolution, step/mm)
$101 = 320.000    (Y-axis travel resolution, step/mm)
$102 = 320.000    (Z-axis travel resolution, step/mm)
$110 = 2800.000    (X-axis maximum rate, mm/min)
$111 = 2800.000    (Y-axis maximum rate, mm/min)
$112 = 2800.000    (Z-axis maximum rate, mm/min)
$120 = 150.000    (X-axis acceleration, mm/sec^2)
$121 = 150.000    (Y-axis acceleration, mm/sec^2)
$122 = 150.000    (Z-axis acceleration, mm/sec^2)
$130 = 200.000    (X-axis maximum travel, millimeters)
$131 = 300.000    (Y-axis maximum travel, millimeters)
$132 = 50.000    (Z-axis maximum travel, millimeters)

The test code:

T1
G17
G21
G90
G0Z6.0
G0X0Y0S12000M3
G0X9.0Y-4.0Z6.0
G1Z-1.6000F60.0
G1X10.0Y0.0
G1Y20.0
G1X15.0Y25.0
G1Y35.0
G0Z6.0
G0X0Y0
M30

That code is sent with UGS

swarfer, 2018-05-21:
why would M30 turn PWM off when M5 is supposed to do that?
your program does not have an M5

Jibeji, 2018-05-21:
Of course I can add it to the end, but the software I use (Cut2D) just sends a M30 command, which is also supposed to stop everything.
As chamnit said, GRBL should act as attended, I surely have a misconfiguration somewhere.

chamnit, 2018-05-21:
Once again tested with a direct connection to Grbl. This is time with your example. Once again works exactly as it should. The spindle disables via M30. As I mentioned in a prior post, GUIs sometimes will scrub certain gcodes, like M30, from your program. Check your UGS settings and try a different GUI like bCNC.

Jibeji, 2018-05-21:
Just to be sure, when you say:

> The spindle disables via M30

Is it because the PWM signal goes to 0 or by using Spindle Enable pin?

chamnit, 2018-05-21:
@Jibeji : You stated that you were using the standard build of Grbl. This build does not have a spindle enable pin. Only PWM on D11 and spindle direction on D13. PWM turns off when the spindle is disabled.

Jibeji, 2018-05-21:
OK, I was asking because there is a SP EN pin on the documentation, that I do not use.
Thanks for your tests, I'll try with another sender.

chamnit, 2018-05-21:
It’s likely UGS scrubbing some gcodes. Can you try a different sender? Or the Grb

chamnit, 2018-05-21:
Ignore last comment. Fat fingers.

closed 2018-05-21 by chamnit