gnea/grbl #445: clearification about system_???_exec_????_overrides

cri-s, 2018-05-20:
Hi, looking over the code, enabling most of the features (N line, coreXY, ...) 
it compiles over 32k on avr target.

Now looking over the code i have seen this type of functions that just update volatile 8bit variable.

void system_clear_exec_motion_overrides() {
  uint8_t sreg = SREG;
  cli();
  sys_rt_exec_motion_override = 0;
  SREG = sreg;
}

As the veriable is volatile and atomar, what is the reason for this function and interrupt blocking overhead ?

Regards
Cri.s

chamnit, 2018-05-20:
I believe it was to prevent the serial interrupt from firing and setting this variable while clearing this variable.

closed 2018-05-20 by chamnit

langwadt, 2018-05-20:
with all the functions that are just a write and not a read/modify it shouldn't be needed, the write can't be interrupted so a serial interrupt will happen before or after regardless

chamnit, 2018-05-20:
@langwadt : Yeah probably, but it doesn't hurt for it to be there. Mostly defensive programming. Either way, removing it won't same more than a few bytes of flash and a handful of cycles since sei() and cli() are compiled into a single assembly instruction.

As a side note, I did find a way to free up about a 100 bytes of flash or so by altering how the default settings are copied. Not much, but it's still best to just go to the Mega2560 version if you really need the extra space.

cri-s, 2018-05-20:
Actually I’m on 102.4% flash for uno and need maybe 0.2% to add. Without
full options it is on 95%.
The biggest surprise was ram .
Changing 32 to 24 bit changed it from over 105% to 75%. This for 6 axis .
Maybe im using the wrong fw as there are uno and mega versions.
A big surprise was the new M56 opcode as this Is normally used basically as
relay output.

langwadt, 2018-05-21:
@chamnit I was curious so I tried changing the four functions that are just a write, it saves 30 bytes

chamnit, 2018-05-21:
@langwadt : Thanks. 30 bytes isn't a lot. The funny thing is that I don't think I had most of the extra SREG code before, but someone posted here strongly insisting that I put it in. That it was wrong without them. Not sure who is right, but whatever the case, there's no evidence that it's doing anything bad. If it's removed, there will need to be a lot of testing to ensure it doesn't create any new issues.

cri-s, 2018-05-22:
I suggest to remove this inside serial.c

         case CMD_DEBUG_REPORT: {uint8_t sreg = SREG; cli();
bit_true(sys_rt_exec_debug,EXEC_DEBUG_REPORT); SREG = sreg;} break;

and replace it with
         case CMD_DEBUG_REPORT:
bit_true(sys_rt_exec_debug,EXEC_DEBUG_REPORT); break;

As the code is inside interrupt code, global interrupt bit is are
always cleared,
Further this increase portability removing processor dependent special
code outside
defines.