gnea/grbl #446: In LaserMode, turning off the laser also shortly stops the move

DanielWeigl, 2018-05-20:
After reading the documention for the lasermode, I wanted to make a calibartion-pattern to see how `M3` vs `M4` affects the outcome, I made a script to generate the pattern: https://jsfiddle.net/0r4yb01a/

I noticed, that as soon as i try to make rectangles (switching off the laser for 2mm after each `S` setting), the motion isnt smooth any longer.

So, this is smooth (no stops between `S` changes):
```
$32=1
G90
M4 S0
F1800
G1 S0
G0 X0 Y0
G1 X0 Y0 S10
G1 X3 Y0 S10
G1 X5 Y0 S30
G1 X8 Y0 S30
G1 X10 Y0 S50
G1 X13 Y0 S50
G1 X15 Y0 S70
G1 X18 Y0 S70
...
```

but as soon as i try to switch off the laser, it also pauses for a short period of time between switching off (or switching on?)
```
$32=1
G90
M4 S0
F1800
G0 X0 Y0
G1 X0 Y0 S10
G1 X3 Y0 S10
G1 S0
G1 X5 Y0
G1 X5 Y0 S30
G1 X8 Y0 S30
G1 S0
G1 X10 Y0
G1 X10 Y0 S50
G1 X13 Y0 S50
G1 S0
G1 X15 Y0
G1 X15 Y0 S70
G1 X18 Y0 S70
...
```

I also did not see any difference between `M3` and `M4`, also tried not completely switching to `S0`, but eg `S50`, then I noticed the first moves are smooth, but as soon as the S value is lowered, it slows down. Could this be a problem? Or did I understand something wrong about the Laser-Mode?

```
>>> $I
[VER:1.1f.20170801:]
[OPT:V,15,128]
ok
>>> $G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
```

(XCarve with XController, Sender: tested it with UGS and bCNC, no difference)

chamnit, 2018-05-20:
Remove the `G1 S0` lines and just append an `S0` to the line after. A G1 without any motion attached to it will force Grbl to sync and pause, because it has to ensure that the laser is set at the correct time, which is after all of the motions in the planner are executed.

closed 2018-05-20 by chamnit

DanielWeigl, 2018-05-21:
Thx for the quick response, yes you are right, now it works . I tried a lot of different combinations, but it seems this one I missed.

DanielWeigl, 2018-05-21:
...But not completely: I tried your suggesion ( https://jsfiddle.net/DanielWeigl/y2wufzba/4/ ) with `M4`-> this runs smooth, but same G-code with `M3` again stops on `S0`  -> is this intended?

chamnit, 2018-05-21:
You have motions that program coincident targets sequentially. This is also equivalent to no motion programmed.