gnea/grbl #451: New grbl v1.1 config issue

brucect, 2018-05-27:
I have uploaded grbl v1.1 and when I go to the config file I notice it just says config no .h and when I change a line and go to save it it asks me where I want to save it. Never seen that before either. Anyone, can you help??
Thanks

lucky62, 2018-05-27:
Most probably  you need to change the files permissions. Are you on Linux?

brucect, 2018-05-27:
No. My operating system is Windows 10.

lucky62, 2018-05-27:
Check the your effective rights on the file you are trying to edit. In Windows it is slightly complicated due to rights inheritance. File editor is asking for place usually when file is read only or if you cannot modify the file due to insufficient permissions/rights.

idorual, 2018-05-27:
Hello!
Excuse I have to use the translator, I speak french.
If that can help you !!! I have windows 7 and when I modify a file inside the directory C: \ Program Files (x86) it always asks me to type the password of the user 'Administrator before giving me the permission to modify a file . This is because I installed a password to open windows under the user 'Admininistrateur', if I could enter the user 'Administrator' without password I could not modify a file in this directory.
I hope that this answer will give you an idea of the solution of your problem on Windows10.
idorual

lucky62, 2018-05-27:
@idorual, move the files outside of "Program Files". Best would be to move it to your profile directory e.g. Documents. (C:\Users\<your_name>\Documents)

idorual, 2018-05-27:
Thank Lucky62 
 Excuse me, I wanted to try to explain to brucect why he was not able to edit a grbl configuration file.

I apologize for the translator

brucect, 2018-05-27:
Thank you both for the info. Got it working.

closed 2018-05-27 by chamnit