gnea/grbl #452: GRBL 1.1 with LCD and Keyboard

Chip1972, 2018-05-27:
HI.. I manage to write a 300 bytes routine to put LCD and Keyboard on GRBL 1.1.
I use this circuit:
https://www.pocketmagic.net/wp-content/uploads/2013/02/3wire_LCD_Atmega_new_radu.png
I am reducing the code, but I still need to know what the variables to control XYZ coordinates.
Anyone can help me?

Chip1972, 2018-05-27:
Its possible to use this circuit too, but the code became a bit bigger.
http://3.bp.blogspot.com/-49iYZO34afE/T0tdnbr0l-I/AAAAAAAAADY/WC-r3KG7Qtw/s1600/Serial_LCD_img.jpg
I need help to use the GRBL lib to print XYZ variables in LCD, instead the serial port.

vMeph, 2018-05-27:
you also can add lcd by using a second arduino
https://www.shapeoko.com/wiki/index.php/LCD_on_GRBL#Full_version_GRBL_1.1

chamnit, 2018-05-27:
Your LCD code will grow a lot once you start adding text and other functionality. If you are still persistent in making this work, look at how status reports are generated to see how real-time position is compute.

closed 2018-05-27 by chamnit

Chip1972, 2018-05-28:
It worked, Thanks. Now I will work in the keyboard control.

luisluna2, 2018-09-27:
> you also can add lcd by using a second arduino
> https://www.shapeoko.com/wiki/index.php/LCD_on_GRBL#Full_version_GRBL_1.1

Hello friend, I have interesting in this project but I can't open the page, can send the Master Files by Email? andres.campa_luis@hotmail.com

psyhlo, 2018-09-27:
may be this?
https://web.archive.org/web/20161205051301/http://www.shapeoko.com:80/wiki/index.php/LCD_on_GRBL

luisluna2, 2018-09-27:
> may be this?
> https://web.archive.org/web/20161205051301/http://www.shapeoko.com:80/wiki/index.php/LCD_on_GRBL

Thank you friend, th elink work now, amm I have a question: With this program can use the CNC Machine or is only for check the status in the LCD Screen?