gnea/grbl #455: arduino mega grbl 1.1f stop milling

yuranikspb, 2018-05-29:
`X2.311Y24.088Z-0.035 < ok
X2.947Y24.088Z-0.037 < ok
X3.583Y24.088Z-0.040 < ok
X4.038Y24.088Z-0.041 < ok
G01X4.060Y24.090Z-0.042 < ok
G01X4.075Y24.093Z-0.042
G01X4.089Y24.097Z-0.042
G01X4.102Y24.103Z-0.042
G01X4.115Y24.110Z-0.042
G01X4.128Y24.118Z-0.042
Ошибка последовательного порта 1: No such file or directory
[CTRL+X] < Grbl 1.1f ['$' for help]
[MSG:'$H'|'$X' to unlock]
S1000 < error:9
$X < [MSG:Caution: Unlocked]
ok`
how to solved ?

JBSchueler, 2018-05-29:
Seems your serial port runs into a overrun.
Which serial chip do you have on your arduino?

109JB, 2018-05-29:
What program are you using to send the g-code to Grbl?

yuranikspb, 2018-05-30:
Program: Candle 1.1.8
serial chip: 8u2
cable: "Hama Active USB 2.0 Extension Cable, shielded, black, 5.00 m"
connection speed: 115200