gnea/grbl #457: Control XYZ via Keypad

Chip1972, 2018-05-30:
HI. I manage to add an Keypad to GRBL1.1.
But when I change the gc_state.position[x],  nothing happens to the motor.
when I change the sys_position[x], nothing happens too.
When I call system_execute_line("G91 G0 X1"), the command is not executed.
How move the motor from GRBL code?

Schildkroet, 2018-05-30:
chamnit already addressed your issue. Stop ignoring his advice.

Chip1972, 2018-05-30:
Sorry  Schildkroet, but where you ready anything about LCD? I am not adressing an issue about it. Please respect my question.

chamnit, 2018-05-30:
@Schildkroet is right. I've already answered this question. Once again, don't hack into Grbl. You'll break something. Use the as-designed interface like the jogging commands. If you keep persisting, I will immediately close the issue.

closed 2018-05-30 by chamnit