gnea/grbl #458: Limit switch work, Homing overun?

lquang4321, 2018-05-30:
Using GRBL 1.1 and Arduino CNC shield 3.10

$1=25
$2=0
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=1
$24=25.000
$25=500.000
$26=250
$27=5.000
$30=1000
$31=0
$32=0
$100=754.920
$101=400.000
$102=250.000
$110=1500.000
$111=1000.000
$112=635.000
$120=10.000
$121=10.000
$122=10.000
$130=185.000
$131=240.000
$132=200.000
ok
$23=0 < ok
$$ < $0=10
$1=25
$2=0
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=0
$24=25.000
$25=500.000
$26=250
$27=5.000
$30=1000
$31=0
$32=0
$100=754.920
$101=400.000
$102=250.000
$110=1500.000
$111=1000.000
$112=635.000
$120=10.000
$121=10.000
$122=10.000
$130=185.000
$131=240.000
$132=200.000

I'm making a laser engraver (no laser yet, just testing X&Y). I have disabled the Z axis homing in config.h. When using $H, the X axis begins to move first, so that part works. I'm using hall effect KY-003 (Active low) for limit switch and cube neodymium magnets. They all work for both X & Y if I send G0 commands, stopping for +-X and +-Y. But homing using $H will just overrun the limit switch and won't stop, despite the limit switch gets triggered (Red LED glows if magnet is near). Any possible reason?

chamnit, 2018-05-30:
What do Grbl's status reports say when you actuate the y-limit switch? Anytime Grbl detects a "triggered" limit switch a "Pn:" message will appear in the report. Make sure Grbl is detecting them correctly.

lquang4321, 2018-05-30:
@chamnit Sorry, I was using Candle software and it doesn't have verbose output. After using the universal Gcode, I managed to see what triggers. It outputs that Y switch was triggered when the X axis was moving. I swapped the limit switch and now it stops. Running into another problem though. My Y axis is now missing and jumping steps (no skip). Making it G0 Y10 moves it by 20 cm, when using G0 Y1 moves it 1 mm. I'm using A4988 with ample amount of current(1.4A) for the motor ( 2A/phase max). No clue why.

closed 2018-06-01 by lquang4321