gnea/grbl #462: Mirrored Y axis

yisabell, 2018-06-03:
I have just finished the construction of OpenBuild's ACRO cnc platform and it is driven by a Raspberry Pi with a Protoneer hat, all axis work and are calibrate correctly (in my opinion).  I just discovered that I have a Y axis that is mirrored.  Using bCNC if I do a vertical flip I get the correct results.  This was discovered when printing the letter
[SETTINGS.TXT](https://github.com/gnea/grbl/files/2066152/SETTINGS.TXT)
 S as it was facing the wrong direction.

I played with $2 and $3 as well as $23 but to no avail. Agin in bCNC my work area is in positive X but negative Y.

Any guidance to rectify would be appreciated.

malcolmmp, 2018-06-03:
I'm not familiar with bCNC, but it would be helpful if you posted your config settings (results of $$).

yisabell, 2018-06-03:
I had attached a file called settings.txt, let me know if it did net bet
attached.

Yves Isabelle
905-407-1071
131-2125 Itabashi Way
Burlington Ontario
L7M 0A1

malcolmmp, 2018-06-03:
So do the jog commands work the way you expect them to and does your homing cycle match in software with what you physically see? Does bCNC show the tool location tracking the same way it appears to you?

yisabell, 2018-06-03:
When I jog X & Y  left and right,up & down...yes they match.  Home is
currently in the top righthand corner.with $3=2 and $2=0.

And yes if I load a file the tool (in this case a Laser) track the same way
as the jogs.  but its inverted, such as in the letter S is flipped
horizontally.

For some reason my workarea is in the bottom righthand quadrant and I cant
seem to make it the top righthand.

Any direction would be appreciated.

Thanks in advance.

Yves Isabelle
905-407-1071
131-2125 Itabashi Way
Burlington Ontario
L7M 0A1

yisabell, 2018-06-04:
Ok so I have set $2=2 and $3=2 and the result is correct however every file I load up I have to pull it down from the top right-hand quadrant to bottom right-hand quadrant.  I am curious why it seems to be locked in to the bottom.