gnea/grbl #465: Can you solve the incompatibility Problem HOMING_FORCE_SET_ORIGIN vs. PARKING_ENABLE?

ml-hn, 2018-06-04:
[HOMING_FORCE_SET_ORIGIN is not supported with PARKING_ENABLE at this time.](https://github.com/gnea/grbl/blob/master/grbl/grbl.h#L85)

I use a positive X/Y space, but I also need the parking option for the Z axis. I hope and think that there should be a simple solution. Of course, only if someone knows how.

chamnit, 2018-06-04:
It's not supported as it states. It was mainly due to space constraints of having to check and calculate all of the different possibilities. It was more efficient to write it in terms of a standard machine setup. I suggest that you revert back to the standard machine coordinate system with all negative space and use work coordinate systems to set your coordinate frame to how you want.

closed 2018-06-04 by chamnit

ml-hn, 2018-06-04:
Hi,
I use a Atmega 644 with enougt space.

chamnit, 2018-06-04:
@ml-hn : That's great, but the 644 isn't officially supported. If you want to add this ability, you'll have to edit the parking source code to make it do what you want.

ml-hn, 2018-06-04:
Ok, then I'll get into C programming myself. Unfortunately, I have another problem. The incompatibility of the 644 with [this](https://github.com/gnea/grbl/blob/master/grbl/grbl.h#L76) option.

Thank you!