gnea/grbl #472: Homing procedure fails when larger max travel values are set

mikemaikaefer, 2018-06-09:
I have configured GRBLs max travel distances ($130, $131, $132) to quite large values because the controlling software manages these limits. But this lets the homing procedure fail, when I do e.g. a command $HX, all step-outputs get only one short pulse (yes, all after each other, like some kind of lightshow) and that's it. GRBL blocks, does not move any axis and does not react on the limit/home switches.

This happens with all travel values larger than 150000.000, a value of 110000.000 seems to be safe. So my question: is this some kind of known limitation? If yes: what is the maximum value allowed here? If no: any bug?

langwadt, 2018-06-09:
32 bit integer is ~ +/- 2e9, so that would be good for ~13000 step/mm at 150meter

chamnit, 2018-06-09:
Oops. Forgot a few orders of magnitude. Either way the travel limits are crazy high. Homing uses those travel limits to determine how far to search. It’s more likely the high values are truncating some floating point operations. Either way reduce the value to something more reasonable.

Elmi77, 2018-06-11:
OK, I fully understand there are limits for these values. But instead of showing an undefined behaviour during homing (but not during movement!) which can't be brought into relation with this value easily (it took me 3 days to find this out), wouldn't it be better to limit the values internally or to give an error when a too high value is set?

chamnit, 2018-06-11:
Yes it would if there was space for checks like this. Grbl has literally a few dozen bytes of flash left for some critical build configurations. I intend to do a lot more checks and provide feedback in the next version when there is a lot more room on an ARM chip.