gnea/grbl #473: Override question. Not necessarily a bug

109JB, 2018-06-09:
I finally have a chance to do some work updating my GUI and had a question. It may or may not be a bug.

So lets say you have a programmed feed rate of 100 mm/min and set the feed override at 80%. In this case, Grbl reports the feed rate as 80 mm/min.

Now lets say you have the spindle speed programmed at 1000 rpm and the spindle override at 80%.  Grbl still reports the spindle speed as 1000 rpm. I would have expected it to report 800 rpm.  Is this a bug or is it intentional?

chamnit, 2018-06-09:
Probably a bug. I may have forgotten to do that conversion in the report.

chamnit, 2018-06-09:
I can verify that this is a bug. It doesn't update the spindle override when idle. Once motion is active, it will update.

chamnit, 2018-06-09:
This is a duplicate of #268. Closing.

closed 2018-06-09 by chamnit