gnea/grbl #474: Ading GRBL library to Arduino IDE for compiling GRBL

HuubBuis, 2018-06-09:
When i follow these instruction (IDE version 1.8.5) for adding the GRBL library to the arduino IDE

"**_IMPORTANT: Select the grbl folder inside the grbl-XXX folder, which only contains the source files and an example directory_**"

I get an error message "**_A subfolder of your sketchbook is not a valid library_**", the library is not added and I can not compile GRBL. When I make a zip file of this folder, all is working well.

chamnit, 2018-06-09:
Tested this on 1.8.5. Worked fine. No errors.

closed 2018-06-09 by chamnit

109JB, 2018-06-10:
Don't add the library from within the idea.  Just copy the grbl folder into the libraries folder using whatever file Management program you use (ie: windows explorer)

HuubBuis, 2021-05-08:
You should be able to compile grbl using the steps described in the [WiKi](https://github.com/gnea/grbl/wiki/Compiling-Grbl)
or use the method from 109JB above. 

Before you add the library, you have to remove all (old) grbl directory(s) in your sketch folder map **libraries**
There may also be conflicting libraries. 
You could start by using a (new) clean sketch map. After removing all old libraries, you have to restart the Arduino IDE.