gnea/grbl #475: Issue with turning off mist and flood coolant

109JB, 2018-06-11:
Working on my GUI and I found that when both mist and flood coolant are enabled and both turned on, they cannot be turned off until both a 0xA0 and a 0xA1 are issued. Also, if the sending of the 0xA0 and 0xA1 commands aren't done right they are ineffective. If turning mist or flood on individually without the other turned on everything works fine. The problem occurs when both are turned on at the same time. 
Note:  I didn't have my meter handy, so I can't say if this is just the reporting and the pins are functioning correctly, or if if the pins are also not toggling.

Here is a run of status reports with inserts where the toggles were sent to show the problem and some notes.

### First section shows that using flood and mist individually works as expected

<Idle|MPos:0.000,0.000,0.000|FS:0,0>
>>>> 0xA1 (Mist toggle) sent
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100|A:M>			Mist on
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:2.000,2.000,0.000>
>>>> 0xA1 (Mist toggle) sent
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>				Mist off (no problem as long as flood isn't also on)
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:2.000,2.000,0.000>
>>>> 0xA0 (Flood toggle) sent
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100|A:F>			Flood on
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
>>>> 0xA0 (Flood toggle) sent
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>				Flood off (no problem as long as mist isn't also on

### This next section shows the problem

<Idle|MPos:0.000,0.000,0.000|FS:0,0>
>>>> 0xA1 (Mist toggle) sent
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100|A:M>				Mist turned on (No problem so far)
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:2.000,2.000,0.000>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100|A:M>
>>>> 0xA0 (Flood toggle) sent
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100|A:FM>			Flood and mist both turned on (Again, no problem so far)
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:2.000,2.000,0.000>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
>>>> 0xA0 (Flood toggle) sent
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100|A:FM>			Flood toggle sent but it remains on (Should have turned off flood)
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:2.000,2.000,0.000>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
>>>> 0xA0 (Flood toggle) sent
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100|A:FM>			Flood toggle sent but it remains on (Still won't turn off flood)
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:2.000,2.000,0.000>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
>>>> 0xA0 (Flood toggle) sent
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100|A:FM>			Flood toggle sent but it remains on (Still won't turn off flood)
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:2.000,2.000,0.000>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100|A:FM>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
>>>> 0xA0 (Flood toggle) sent
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100|A:FM>			Flood toggle sent but it remains on (Still won't turn off flood)
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:2.000,2.000,0.000>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100|A:FM>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:2.000,2.000,0.000>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
>>>> 0xA1 (Mist toggle) sent
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100|A:FM>			Mist toggle sent but both remain on (Should have turned mist off)
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:2.000,2.000,0.000>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100|A:FM>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
>>>> 0xA1 (Mist toggle) sent
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100|A:FM>			Mist toggle sent but both remain on	(Still won't turn off mist)
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
>>>> 0xA1 (Mist toggle) sent
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100|A:FM>			Mist toggle sent but both remain on (Still won't turn off mist)
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:2.000,2.000,0.000>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100|A:FM>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:2.000,2.000,0.000>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
>>>> 0xA0 (Flood toggle) sent
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>					Flood toggle sent again and now both turned off (Only should have turned off Flood)
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>

In the above, note that the flood toggle was tried 4 times before the mist and the mist tried 3 times. If I had done the mist 4 times then the last flood toggle still wouldn't have turned things off.

chamnit, 2018-06-11:
Ok thanks. Definitely a bug. Fortunately, I've been working on the other spindle override bug and they are in close proximity to each other.  So, I'll be able to test and fix these at the same time.

chamnit, 2018-06-11:
@109JB : Thinking about this a little more. The g-code standard doesn't have a toggle for either coolant. An M9 disable coolant command disables both M7 and M8, regardless if one or both are enabled. I suppose the question would be: should the override toggles behave the same way? Or allow you to disable one or the other separately? I'm leaning toward the latter.

109JB, 2018-06-11:
I agree.  Although in G-code the M9 turns both off, I see no reason that the toggles within Grbl need to act the same way. I personally would prefer that the Grbl toggles act individually. 

Additionally I can see using one or the other for something different from coolant, in which case having the toggles act independent would be beneficial.

chamnit, 2018-06-14:
@109JB : Just pushed a fix. Should work, but oplease try to break it. Let me know how it goes.

closed 2018-06-14 by chamnit