gnea/grbl #477: Feed Rate adjustment when running

einencool, 2018-06-14:
Hello,
I‘m using the Software BCNC with a Arduino Mega and Grbl 1.1f 
Everything works fine, and I‘ve a little Num Pad to move the Axis and to start/stop a program.
In the Software there is the ability to execute shortcuts. 
So I would like to adjust the feed rates while the program is running.
The wiki says the following:
0x90 : Set 100% of programmed rate.
0x91 : Increase 10%
0x92 : Decrease 10%

But I don‘t know how to enter this codes, so that the feed rates will change.
Can you please help me?
Thank you in advance
Greets Chris

109JB, 2018-06-14:
They are extended ASCII codes. You can't really just send them like text. Your GUI  has to be set up to send them for you through a button or slider or something like that that you can interact with. In your case, with bcnc, there is a slide control marked "Feed override". Use that.

einencool, 2018-06-15:
Thank you for your fast reply.
Normally I use that slider, but it is not do easy because the computer is not directly at the cnc..
Therefore I would like to use shortcuts and send them with the num pad. 
I thought there would be an option to execute this.
But when this is not an option, then I have to live with it...

On the num pad I have the function to jog the gantry, home it, start stop a program and to zero the z-axis. All this is routed through the shortcuts.

But thank you for your fast response.
Greets Chris

einencool, 2018-06-24:
Hi chamnit,
sorry for the late reply.
With sendhex 21 it works, and the machine stops, but with the sendhex 144 I can't change the feed rates.
I posted this in the BCNC github, but i don't know where is the error.
[https://github.com/vlachoudis/bCNC/issues/856](url)

Thank you 
Greets Chris