gnea/grbl #485: grbl via bluetooth hc-05 speed problem

yuranikspb, 2018-07-06:
Hi everybody.

I made a connection to cnc through two modules hc-05 (master-> slave)
But there was a problem of the sneakiness of the transfer

the transmission speed via bluetooth is not more than 9600 approximately
settings are all set correctly hc 05 is set to 38400
(other values ​​of the data transfer rate also do not matter)

I found an article that is similar to my problem
https://forum.arduino.cc/index.php?topic=424791.15

   > "You are encountering the age old problem of latency. A synchronous packet system cannot send each of your asynchronous bytes without imbedding them in a packet"

what can i do to low down latency.
or use another module?

zeevy, 2018-07-06:
Why dont you use only one bluetooth with android application?

yuranikspb, 2018-07-09:
I want to work with pc.

yuranikspb, 2018-07-09:
I test with  with my laptop via bluetooth, and got the same result

zeevy, 2018-07-11:
Check your laptop com port settings, and its max speed

yuranikspb, 2018-07-11:
i  change com port speed  in windows device manager. no effect
I connect grbl via wifi esp 07  and get good speed! but many errors :(error:1 error:2 error 23)

109JB, 2018-07-11:
I tried an HC-05 several years ago, and found the throughput speed to be pretty slow. I figured it would be OK for general milling or routing, but for anything like 3d milling, laser work, etc, that requires high speed it would not be suited. I was never able to get the speed up and abandoned the idea.

langwadt, 2018-07-11:
is it a latency issue with buffers getting starved waiting for acknowledgment? When I worked on the very first Bluetooth nearly 20 years ago we had no problem maintaining the theoretical max of 721kbit/sec

109JB, 2018-07-12:
Yeah. I've seen the uno with the built in esp8266. I don't have one of those but I do have esp 8266 boards and was contemplating seeing if I could connect one up to an Arduino.  May have to give that a try.

xreef, 2018-07-13:
I'm using HC-05 bluetooth module on my Homemade Router with CNC Shield, and work correctly, you must set 112500 baud rate to you Bluetooth module.
Look here for configuration of Bluetooth https://www.instructables.com/id/Arduino-Remotewireless-Programming-and-Test-With-H/ .

109JB, 2018-07-14:
> I'm using HC-05 bluetooth module on my Homemade Router with CNC Shield, and work correctly, you must set 112500 baud rate to you Bluetooth module.

Mine was set to 115200 baud. It worked reliably, just too slow for what I wanted. If baud rate wasn't right it wouldn't have worked at all.

As I said when I tried the HC-05, I got it working just fine, but when running throughput intensive programs, such as 3d contour milling code, the Grbl buffer was starving quite often. Running the exact same program with a USB hookup and the GRBL buffer remained filled most of the time.

For 2.5d milling or programs with similar throughput needs, I'm sure it would be fine.

zeevy, 2018-07-14:
To overcome the problem of grbl buffer straving with bluetooth, we can increase the serial rx buffer, say 192. (the gcode sender you are using should be aware of this, and use that value for buffered streaming). 
I have achived good results on my machine with this change.

HuubBuis, 2018-07-14:
<Running the exact same program with a USB hookup and the GRBL buffer remained filled most of the 
time>
If have tested several Bluetooth modules (HC-04, HC-05, SPP-C) and the HC-05 module was the fastest. The ESP8266 WiFi module sends commands 15 times faster than the Bluetooth HC-05 module.  It could solve your problem without changing GRBL. Give it a try!

yuranikspb, 2018-07-16:
> ESP8266 WiFi module sends commands 15 times faster than the Bluetooth HC-05 module. It could solve your problem without changing GRBL. Give it a try!

yes, but  more garbage :(

HuubBuis, 2018-07-16:
> yes, but more garbage > 

What do you mean by more garbage?

yuranikspb, 2018-07-16:
hc-05 work good, no garbage via serial port, but slow.  esp8266  work fast  but grabage via serial port and grbl stop

HuubBuis, 2018-07-16:
I am using GRBL and ESP8266 modules over a year. No problems at all, works reliable in my 4 x 8 meter shop.

There are ESP8266 modules with builtin USB connection, no extra wiring needed. Just connect the USB and you are done!

NodeMcu Lua ESP8266 ESP-12E

hariharan382, 2021-04-18:
can you say how you connected hc05 via pc ugs,bcnc?

xreef, 2021-04-18:
I think this can help 
https://www.mischianti.org/2019/05/15/cnc-cyclone-pcb-factory-control-buttons-and-bluetooth-part-5/

HuubBuis, 2021-04-18:
> 
> 
> can you say how you connected hc05 via pc ugs,bcnc?

I am not familiar using bcnc but UGS only supports serial or ethernet (WiFi).
[https://winder.github.io/ugs_website/guide/platform/#connecting-to-the-controller](url)

HuubBuis, 2021-04-18:
> 
> 
> can you say how you connected hc05 via pc ugs,bcnc?

As an alternative you could use a serial to bluetooth connection on the PC and let the gcode senders use the serial port.
The grbl controller could be connected the same way using a Bluetooth to serial adapter. The two bluetooth modules should connect to each other. 
Checkout this [Link](https://howtomechatronics.com/tutorials/arduino/how-to-configure-pair-two-hc-05-bluetooth-module-master-slave-commands/):

hariharan382, 2021-04-22:
what about laptop bluetooth ,should i have to go for another bluetooth?
i tried in ugs it is not working but trying it in grbl controller but it too shows me error

HuubBuis, 2021-04-22:
> what about laptop bluetooth

UGS can't communicate using the internal laptop Bluetooth module. An external serial connected Bluetooth module will be seen by UGS as a serial port.