gnea/grbl #488: Weird glitches with movement accuracy

errno, 2018-07-14:
I have a pretty big MPCNC based laser engraver/cutter with GRBL 1.1f (custom built for disabling Z axis homing). But I have an issue with movement accuracy, not straight lines, and shifts. 
[image]
I've eliminated skipped steps as the cause, kind of. There is no binding in kinematics, no indicative sound. And this is 100% repeatable, I did run the same G code several times in a row without moving a material - 100% match. I think, skipped steps have to be sporadic or at least slightly deviating.
I've seen mentions of driver's temperature cutoffs as a possible cause. Indeed, drivers were pretty hot, but reducing current and adding active cooling (till heatsinks are barely noticeable warm) didn't solve the issue.
I've tried to reduce a feed rate down to 300mm/min without any difference, still 100% the same result.
Offsetting a drawing to another part of the work area (to ensure that this is not kinematics binding) showed exactly the same defects.

I'm frustrated. Any ideas? What diagnostic steps I've missed?

P.S. I've attached example drawing and G code.
https://www.dropbox.com/s/3r71mmzgqvi6fq0/Dimond_test.svg?dl=0
https://www.dropbox.com/s/gnbschb8eyzmwh6/dimond_test2a.gcode?dl=0

chamnit, 2018-07-14:
This looks like either a machine, streaming, or settings issue. Can you post your Grbl $$ settings, $I build info strings, and what you are using to stream to Grbl?

errno, 2018-07-14:
Sure, here are the settings and build info:
```
$I
[VER:1.1f.20170801:]
[OPT:V,15,128]
ok
$$
$0=10
$1=25
$2=0
$3=0
$4=0
$5=1
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=1
$21=0
$22=1
$23=5
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=255
$31=0
$32=1
$100=100.000
$101=100.000
$102=100.000
$110=1000.000
$111=1000.000
$112=1000.000
$120=10.000
$121=10.000
$122=10.000
$130=1101.000
$131=801.000
$132=200.000
ok
```
I'm using GRBL Controller on Win10. Serial connection speed 115200.

chamnit, 2018-07-14:
Nothing out of the ordinary in terms of settings. Have you tried a different sender like UGS or bCNC? With problems like this one, i always suggest disconnecting everything, test, and add one component at a time and retest. It usually is either a mechanical problem or related to EMI.

errno, 2018-07-14:
Thanks for the suggestion.
The machine is quite big, so I had to use a USB extension to reach my desktop computer. That might be an issue.
I'm gonna try a Raspberry Pi connected with a short cable and other senders. I'll update/close this issue after that.
I appreciate your help.

neilferreri, 2018-07-15:
@errno Can you run the same test at a different location in your work envelope?  That looks mechanical to me, but it might help to test in a different spot.
Can you give more info about the electronics?  What steppers, drivers, power supply, etc?