gnea/grbl #489: Individual Stepper Unlock (Z axis)

JesseFo, 2018-07-17:
I have a setup where the Z axis is attached to a stepper, but the vast majority of the time is not in use. In fact; the Z axis is never used while the X and Y axises are engaged. The stepper lock on Z is causing the motor to overheat.

Due to the fact that Z is not used at all while X and Y are, I would like to unlock the Z motor.
I'm using the CNC Shield. Switching boards is potentially an option.

Is this functionality at all possible with grbl?

svenhb, 2018-07-18:
All drivers are getting the same enable signal from grbl - you need an other pin to get a 2nd enable-signal, then you need a new CNC shield to get the new signal to your Z-driver . 
Why not reducing the current to avoid overheating?

Schildkroet, 2018-07-18:
If your stepper is overheating, you don't have a enable problem. You have a problem with wrong hardware setup!

109JB, 2018-07-19:
Many people wrongly believe that a stepper is too hot simply because it feels hot to the touch. Most steppers are rated for case temperatures upwards of 100 deg C (212 deg F), which will burn you in just a few seconds of contact. The way to tell if your steppers are overheating is to use a infrared thermometer and actually measure the case temp. I think you will find it well below the max that the stepper is rated for.