gnea/grbl #496: Enhancement -  Disable Z axis so homing works when there are only two axis in use.

RWL-69, 2018-07-28:
As I am new to GRBL as I want to use a laser etcher for making circuit boards.  I have a Chinese laser and using a Mega as a controller.  I installed limit switches for safety but when I tried to enable homing, I keep getting errors because there is no Z axis.

I have read some lists that recommend installing a switch to manually fake a limit switch engaging.  I think it would be better if there was a way to ignore the Z axis if there is no Z motor using some flag or modify the homing $22=X  where X is a different mode.

1 - Normal XYZ axis
2 - 2 Axis XY

This may be useful for future when GRBL is capable of handling more axis.

biasedlogic, 2018-07-28:
Please, read the wiki and search this issue tracker. This has been addressed multiple times, the configuration is there in the config.h file

RWL-69, 2018-07-28:
Sorry it was late and I was tired.  I read the wiki and there was no mention that I could find.  I also looked through the issues last night before posting.   Didn't see it.

I will search closer now.

closed 2018-07-28 by RWL-69