gnea/grbl #498: Question abiut using a Dremel 3000

dansperfect, 2018-07-30:
So im using a chinese cnc. The spindle that comes with it works good for most thing can even do aluminum. But i want to use my dremel with the board. This may sound stupid and i think i know the answer but Since my dremel is wall powered and the other spindle is being controled by the control board do i just replace my spindle with the dremel and manually control speed and such for the dremel or is there a way for me to setup the board to control the dremel?

chamnit, 2018-07-30:
From what I understand, you can use a solid state relay to turn your Dremel on and off. Speed control may not be possible on a Dremel.

closed 2018-07-30 by chamnit

dansperfect, 2018-07-30:
Awesome i can build that thank you u very much