gnea/grbl #500: GRBL crash/flood with illegal responses

Elmi77, 2018-08-06:
Randomly I notice a behaviour which looks like some kind of crash for me: after some time, GRBL does not send back a valid response any longer but floods my client application by sending huge amounts of ASCII-character "4" - until I do an hard reset (Ctrl-X does not work in this case)

I'm working with main trunk head from GIT on a standard ArduinoNano. What I guess at the moment what causes these problems: it seems to depend on my method of sending commands to GRBL. So instead of sending e.g.

$110=somevalue\n
$120=somevalue\n
$111=somevalue\n
$121=somevalue\n
G0X20.000Y30.000\n
?

with each line sent separately, I submit something like

$110=somevalue\n$120=somevalue\n$111=somevalue\n$121=somevalue\nG0X20.000Y30.000\n?

with all commands sent at once. Here I always keep care of GRBLs buffer, so in no case more than 128 bytes are sent and in no case the commands are sent too fast to have an buffer overflow.

I also tried to work at lower baud rates (recompiled GRBL for that) but this does not make a difference.

What could be the reason for this? Isn't the code from GIT a stable version? Or is it a problem in UART receiving of GRBL which can't handle data when they are sent too fast?

chamnit, 2018-08-06:
This is an EEPROM write issue.m, inherent to the AVR. See this [note](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#eeprom-issues)  to see how to work around it.

closed 2018-08-06 by chamnit

Elmi77, 2018-08-06:
OK, I do not need these data to be written into EEPROM, since I send them down to the Arduino every time I start the application. So is there a way to disable EEPROM-writing and to keep these values in RAM instead?

chamnit, 2018-08-07:
Grbl was built with the assumption that a user won't be changing the EEPROM settings very often, let alone every time an application connects. EEPROM also has finite write cycles. It's possible that the repetitive writes you've been doing has started to make your EEPROM unstable. 

There is no current way to define these settings values without writing to EEPROM. In the next version, it will.