gnea/grbl #501: z axis lead screw falling

hd-pulse, 2018-08-06:
hi 

i have a problem, when my cnc is working how can i lock z stepper motor when its not moving, i want to fix the z axis to not falling down!

i dont want to use pulling spring or counter load, like marlin firmware in prusa i3 3d printer.

thanks.

biasedlogic, 2018-08-06:
Afair you can't do that per axis, only for all axes at once. Set the
timeout to 255
Iirc:
$1=255

hd-pulse, 2018-08-06:
thanks, but i want to know how can i do it electrically or by changing some part of grbl code.

comps, 2018-08-06:
"Electrically", the disabling is done by pulling up the "Stepper Enable" (EN) pin of each stepper driver. If you set `$1=255`, grbl will remember it and will always keep the pin low (no need to change any code). You can achieve the same effect by soldering off the EN pin on each (or just the Z) drivers and wiring it to logic GND. Finally, grbl itself cannot keep only some drivers enabled as it outputs only one pin, "Stepper Enable" (see [the wiki](https://github.com/gnea/grbl/wiki/Connecting-Grbl)) that is connected to all drivers.

As a disclaimer; see the datasheet for your particular driver.

hd-pulse, 2018-08-06:
thanks, as you know disabling stepper makes it free to move and not lock it, i'm thinking of a way, after 50 milisec (less or more) of last step z pulse from arduino (in a single z axis move) by some circuit lock the z stepper and by new pulse before moving release the stepper. 

i think its the hard way, the easy way might be changing the code and its neat!

chamnit, 2018-08-06:
@hd-pulse : Set your $1 setting to 255. This will keep your steppers enabled through the step enable pin whenever Grbl is idle. Please read the Grbl configuration page on this setting. It'll be more clear if you do.

closed 2018-08-06 by chamnit

hd-pulse, 2018-08-06:
yes i read it but $1=255  will impress all the steppers not just z axis alone.

idorual, 2018-08-06:
Hello hd-pulse
For the effect you are looking for on the z axis, you will find it with $ 1
= 255.
Regarding the other axes, I do not see why this would cause a problem if
they keep their stopping position, unless the current is well adjusted to
the standard for these stepper motors

Excuse me for google translation  french to english 😐

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Garanti
sans virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

hd-pulse, 2018-08-06:
okay, thanks idorual, thanks everyone, i do appreciate every moment you guys spend on this topic.

109JB, 2018-08-06:
If you want the Z-axis to stay enable all the time, but the X and Y to disable Per $1, then on most stepper drivers if you just unhook the enable wire from the driver then the drive is enabled all the time. Do this on the Z-axis and then you can set $1 however you want for x and y