gnea/grbl #503: GRBL crash/flood with illegal responses also when using no EEPROM code

Elmi77, 2018-08-07:
I don't understand why my bugs are always closed before I can react on them. So this is a follow-up to bugs #500 and #502:

Also when using absolutely no EEPROM commands but only my "G0"-sequences, GRBL dies after some time, again with sending a lot of trahs via the serial interface.
And this for sure is not a EEPROM-failure, it is specified to work for at least 100000 write cycles and I'm far below of this.

So could this be a issue in current GIT-software since I do use it right from there and not any official release? Or what else could be wrong?

chamnit, 2018-08-07:
Because they were duplicates of the same issue that involved a non-recommended way of streaming settings data. You can still post replies in closed issues and will be re-opened when required. And, yes, EEPROM has a lot of write cycles, but it's quite easy to reach thousands of write cycles with an accidentally for/while loop. The 100,000 cycle rating is also not guaranteed. It's just a statistical certainty value, so manufacturing defects can reduce this well below this number.

Here you are now stating that you are seeing random issues without EEPROM writing, which means that this is something different. Grbl has been stable with no reported comm errors that were not externally caused for quite some time. It's way more likely that there are other variables that can be involved. It'd be helpful if you eliminate these like not compiling and installing it in the recommended way in the wiki, using a custom GUI (try bCNC, UGS, or the Grbl stream.py script), using a different host computer with the proper drivers, using an Arduino clone with poor electrical characteristics or bad USB-serial chip known to corrupt serial data, or still having it connected to other electronics (best to do tests with a bare Arduino and Grbl) to remove EMI issues.

cri-s, 2018-08-07:
> The Arduinos are already original ones, so no cheap Chinese clones with even
> more cheap USB-interfaces. I tried two different ArduinoNano which all have
> been programmed as described.
>
> But may be there is something what I'm doing different: I do not use GRBL as
> CNC-device but just to drive some axes with a guaranteed speed limit which
> is defined by the previously programmed $11x-speed-values. So a movement
> sequence in my testing application looks like this
>
> ?$X
> G0X10.000Y11.000Z12.000
> ???????? <wait until end of motion during this time>
> $X
> G0X1.000Y1.000Z1.000
> ???????? <wait until end of motion during this time>
> -> continue at the beginning with all positions +1
>
> The calls to "$X" are just preliminary to disable pending alarms. "?" is
> called from time to time to get and to display the current position.
>
> This may be a bit unusual, but it should be valid commands? Or is it
> forbidden to use only G0?

The ???????? could overflow the emit buffer, if you do it that way,
you should make
the serial send buffer larger.

chamnit, 2018-08-07:
@Elmi77 : Why aren't you using the feed rate F word to define speed? It seems like a weird work around. The $11x max rate values are there to define the absolute max speed of each axes before losing steps under running conditions. Everything should be defined less than that via the feed rate F word. Also, why are you imposing $X commands? Is there something causing alarms to be invoked?

biasedlogic, 2018-08-07:
@chamnit i guess that using the F word will make his axes move at speeds that are hard to determine - his application isn't coordinated motion from what I can understand, but rather three constant speed drives. I wonder if there's something in the integer calculations that might go awry going for extreme distance at speed limit all the time

chamnit, 2018-08-07:
@biasedlogic : Finding the F word value isn’t too difficult. It’s just vector addition.

But you’re right, top speeds could be a problem. @Elm77 what are your $$ Grbl settings and the range of values you are programming?

Elmi77, 2018-08-08:
@biasedlogic this is correct: I do not have a CNC-style device where the three combined axes X, Y and Z have a resulting feed rate. I have three independent axes where I know for each what the maximum allowed speed is and where they have to moved to.  To be honest I have no idea how I should calculate a feed rate out of three maximum speed values and three motion distances...

@chamnit excerpt of my current settings (I tried with different variants without noticing any difference):

$0=10
$1=25
$2=0
$3=0
$4=1
$5=0
$10=1
$11=0.010
$20=0
$21=0
$22=1
$23=0
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=300.000
$101=300.000
$102=300.000
$110=600.000
$111=600.000
$112=600.000
$120=10.000
$121=10.000
$122=10.000
$130=5000.000
$131=5000.000
$132=5000.000

chamnit, 2018-08-08:
@Elmi77 : Regarding settings, the only potential issue I see is the large travel distances. You're pretty close to maximum distances that Grbl can resolve in some of the internal floating point calculations. 

Compute vector F feed rate by this computation: sqrt( dv_x^2 + dv_y^2 + dv_z^2 )
where dv_x = dx/dt or the maximal axis velocity.
dx, dy, dz are simply the delta travel for each axis and dt is the approximate time the motion needs to travel over.

Elmi77, 2018-08-08:
@chamnit I'm using a limit of 5000 mm because I read somewhere the maximum supported by GRBL is 10000 mm, so I added some extra level of security by using halve of it. But: the problem appears far beyond this limit, it happened latest at a distance of 300 mm but sometimes already after 50 mm

Regarding computation of the feed rate: where do I get dt from? I can calculate the time based on constant speed but there is a acceleration phase included I can't really calculate...

chamnit, 2018-08-08:
@Elmi77 : Have you eliminated all of the other variables, like using some other GUI and/or host computer? It could also be serial drivers on your host. 

You can compute dt from multiple sources like where you want to be and how fast you want the axis to get there. dt = dx/vx. 

Acceleration also occurs when you alter the $11x settings. This is essentially a feed rate limit. Internally, Grbl doesn't treat F feed rate and max rate any different.

Elmi77, 2018-08-09:
@chamnit yes, I tried it from within Windows and Linux, additionally also with a direct serial connection which bypasses the USB-chip (in last case problems are even more worse, then sometimes also the response "ok\r\n" is damaged and contains some trash)

Regarding the feed rate: I'll try that. But G0 is a regular G-code command, shouldn't this work too?

ghartc, 2018-08-09:
Can we clarify the range/limit? It seems a limit should be a unit-less
count, unrelated to scale factors...? Unless it's arbitrary.

On Wed, Aug 8, 2018 at 12:10 PM, Elmi77 <notifications@github.com> wrote:

> @chamnit <https://github.com/chamnit> I'm using a limit of 5000 mm
> because I read somewhere the maximum supported by GRBL is 10000 mm, so I
> added some extra level of security by using halve of it. [...]

chamnit, 2018-08-09:
@Elmi77 : Yes, G0 is a regular g-code command, but it generates an internal F feed rate to move that is limited by any one axis max speed. The calculation it uses is the same. Using G1 and a normal F feed rate word gives you more flexibility if you are ok with doing the calculations yourself.

At this point, it's pretty hard to diagnose your problem, as we have seemed to ruled out most things.  If you can create a test scenario that I can reproduce, then I'll be able to tell if it's something related to Grbl source or with its build method. But, you'll have to use the wiki build process, provide all versions (Arduino IDE, Grbl, Windows/Mac, any related drivers), and use either the Grbl stream.py script or a reputable GUI like UGS or bCNC.

109JB, 2018-08-09:
> Have you eliminated all of the other variables, **like using some other GUI** and/or host computer? 

You never answered if you tried a different GUI or not.  Grbl has been working fine for many many people, so the likelyhood that something this basic is a result of a problem in the firmware seems low.  As I see it it is more likely your hardware, or your GUI. MCU's like the Arduino are also very sensitive to electrical noise, so could it be noise causing the arduino to flake?  Also, I have had one occasion where EEPROM was corrupted and caused problems when running even though running was not using EEPROM. The solution there was to reset the EEPROM using the $RST command.

Elmi77, 2018-08-10:
@109JB I do not use any GUI but an own test program. Nevertheless: I check the buffer properly and I send only very few commands now. Also assumed that I'm doing something wrong with my program - GRBL should never hang up this way? When I send things that are not correct I would expect an error or an alarm but not such a complete failure.

EEPROM was flushed/restored/rewritten also for several times, I also tried operate with default parameters only (but still with my sequence of G0-commands).

Btw: what are these "444444444444444444444" I get back - any kind of special error code?

chamnit, 2018-08-10:
@Elmi77: Once again, I won't be able to help you until you can create a test case that I can reproduce on my own. This requires you to use a vetted streaming program to help eliminate some variables. Also please post your Grbl $I build info strings.

109JB, 2018-08-10:
Have you tried it with a bare arduino, different from the one experiencing the problems to eliminate wiring, noise, and a bad arduino?

SailWithChips, 2018-08-11:
I deffinettely recommend try what the masters are asking for... on the other hand and to discard noise options I would try your hardware with your GUI but without ‘power’... just control (3.3 or 5V).
In combination with this I think try your hardware with an standard UGS is a really easy thing to test and a really helpfull way to discard things... good luck!

Elmi77, 2018-08-12:
@109JB @SailWithChips I am using a bare Arduino, noting connected to it (even no LEDs to check the outputs), just the USB-cable. Additionally it is located on the middle of my desktop, far away even from my Laptop.
On the other hand I also tried to use an Arduino without the serial USB chip and have connected with RX/TX directly. Then things are even more worse, Sometimes I get back damaged answers (so instead of "ok\r\n" I receive "oxxxxx\r\n" where x is some trash - but this trash is always the same and does not vary even with a different serial baudrate, so I would guess it is not a problem of electrical interferences)

@chamnit I'm alredy trying to create a test-scenario where this issue can be reproduced easily so that you can work with it - as soon as I have something, I'll post it here. At the moment it looks like "F" and "G1" instead of "G0" does solve the problem - but this is not yet sure since the crash is not that easy to reproduce (sometimes it happens after 50 motion commands, sometimes I send 500 motion commands and everything is fine).

But to repeat my other question: what is this "44444444444444444444444444444444444" I get back in case of a crash - is "4" any special error code?

chamnit, 2018-08-12:
There is no crash code. The fours you are receiving are weird and garbage. I’m unsure why you are getting them, which is why I’ve been asking you to create a scenario that I can test to. 

Another question. What version of the Arduino IDE are you using to compile Grbl? If Windows, are you using the App Store version?

chamnit, 2018-08-12:
Looking back at you Grbl $$ settings, why are you missing some settings? 6, 12, 13, 24, and 25 are not in the list.

Elmi77, 2018-08-12:
@chamnit I compile GRBL using the avr-gcc from Ubuntu 18.04 / make from command line. The missing $-settings have been removed because I think they are not relevant, they all have been at default values.

Btw: you mentioned the next version of GRBL supports parameter setting without the use of EEPROM - where is this development done? Where can one find the related repository?

Thanks for all your efforts!

chamnit, 2018-08-12:
@Elmi77:  When I asked you if you have compiled and flashed Grbl according to the recommended methods, this implied if you compiled and flashed Grbl via the Arduino IDE. I asked this because the Arduino IDE has a particular set of standardized tools that are all the same version. If you're using a different version of avrgcc, you can run into weird problems.

The next Grbl version is not publicly available. It's not going to be until it's ready for beta release, which is soon. There's multiple reasons for this, and i've covered it in other issues threads.

Sortland, 2018-08-13:
@Elmi77 i can confirm GRBL becomes instable under some conditions. i never found out if it is caused by bypassing the USB chip on the arduino as you have tried it, by running it at 20 MHz or if it is simply a bug. my workaround for this: when a received line from GRBL does not contain "error" and does not contain "ALARM" and when it has some text before "\r\n" it must have been an "ok". with this you do not have to struggle with damaged responses. at least for me this is working perfectly

chamnit, 2018-08-13:
I should add that running Grbl at 20 MHz is a custom mod and not a supported build option. 16MHz is standard for Arduinos and running at 20 MHz often requires custom hardware as well as changes to the source code to compile for it. Users doing this can experience issues from a variety of sources.

Sortland, 2018-08-13:
yes for sure - in Makefile there is a parameter CLOCK which has to be changed from 16000000 to 20000000 in this case. but this is obvious, without this modification the USART baudrate would be completely wrong and one could not communicate with GRBL any more. this CLOCK-value is also used for other relevant timing calculations such as the period for the output. so GRBL works with 20 MHz, but the problems i mentioned are completely random. and as mentioned also i'm not sure if it is caused by the 20 MHz clock or the bypassed USB or something else. ignoring the damaged responses is possible and works smoothly here

109JB, 2018-08-14:
@Elmi77 - So again, have you tried it on another Arduino or just the one that gives the errors?

@Sortland - I would add that it is not wise to accept damaged responses. The communication is 2-way and if you are receiving damaged responses at the GUI end, how do you know if Grbl is receiving correct messages on its end. The communication protocol in Grbl is there for this exact reason and in my experience it works perfectly. If you are receiving "damaged" responses from Grbl it is because you have a problem

Sortland, 2018-08-14:
@109jb what do you expect me to do? i'm not familiar with this this avr-stuff so it is out of my possibilities to debug the code and fix the problem. and yes, i already tried different hardware, even completely different atmega-types

109JB, 2018-08-15:
@Sortland - What I am saying is that the grbl communication works. If it didn't there would be many people with issues relating to the communication not working.  There is nothing that you should need to do with regard to Grbl programming.

If you are receiving partial or corrupted messages back from Grbl it is most likely due to either your GUI not handling the messages properly, which should be taken up with the author of that software, or it could be an issue with noise on the line.  First thing anyone using Grbl should do is to have a good USB cable with a ferrite bead which helps with noise reduction and filtering. Also the USB wire should not be run where it can pick up noise from other lines like spindle power lines, stepper motor wires, etc. Then there is shielding of the wiring that will help to reduce the likelyhood of noise introduced on the USB line.

I myself wrote my own GUI  and can attest that Grbl communicates the way it says it will. If it didn't, I would first be looking at the noise issues and my own GUI programming before assuming that Grbl is in error.

Elmi77, 2018-08-20:
@109JB see above, this was already answered (different hardware, no cheap Chinese Arduinos).

@chamnit: Meanwhile I think following sequence can be used to reproduce the problem(s):

- when output enable is low (means no motion currently running), send command G0X10.000Y11.000Z12.000
- as long as enable-output is high (means the last command is working) send a "?" every 50..100 msec to get the current position
- when output enable is low (means no motion currently running), send command G0X0.000Y0.000Z0.000
- as long as enable-output is high (means the last command is working) send a "?" every 50..100 msec to evaluate the current position
- continue at the beginning

This has to be repeated for about 50..500 times in order to see the problem once. I don't know if the timing is relevant but my application reacts quite fast on changes of the output. And that's exactly why I'm using the stepper enable output but not the "Idle" out of the response from "?"-command - the application needs less stupid parsing and can react faster on state changes.

chamnit, 2018-08-20:
@Elmi77 : In your previous post, you state that you compiled by the recommended methods and then stated later that you compiled with Ubuntu and avrgcc. Once again, did you compile Grbl with the avrgcc version provided with the Arduino IDE? 

Just as an FYI, the wiki is user supported. I don’t explicitly test with Linux, only Mac and Windows, so the wiki may outline unsupported instructions. Using the Arduino IDE itself and the recommended example file is the main method everyone should be using. Not the Makefile. 

Again, I can’t help you if you don’t test with something I can reproduce. This includes using a standard GUI, not a custom one.

Elmi77, 2018-08-20:
@chamnit the ArduinoIDE from Ubuntu makes use of exactly the same compiler, so this does not make a difference. Btw: why is a Makefle provided for when it is not allowed to use it directly?

Regarding the GUI: which one do you recommend? Which one is able to implement these cycles of G0-commands dependent on the enable-output?

chamnit, 2018-08-20:
@Elmi77 : With Ubuntu, you have to make sure the environment paths are set to the Arduino IDE avr-gcc directories. If not, they can be using whatever avr-gcc installation on your Linux machine. It's not explicitly clear that this is what you are doing. I'm asking you to use the Arduino IDE compiling method to eliminate variables. I personally use the Makefile method on my Mac, but I also make sure the diff of the firmware is the same between what the Arduino IDE spits out and what the Makefile does.

Don't use G0 dependent commands or try to program the settings midstream. Do something simpler that can be tested. If you still insist on programming settings mid-stream, try using the stream.py script in the doc folder of this repo. You can edit this file to do what you want and post it as a way so I can test it. I can't test some unknown UI program that you have been developing.

Elmi77, 2018-08-20:
Just as you requested: I compiled the whole stuff using the ArduinoIDE (which is incredibly complicated comparing to a plain "make") and the result is _exactly_ the same (I did a binary comparison).

Next: the problem appears exactly in the way and with the sequence I described above. It appears only when using G0-commands repeatedly (why shouldn't I use them? they are legal G-code commands!) and I do not change settings midstream (please refer to my description above, all I am doing is sending a ? from time to time and checking the enable-pin to see if motion has finished).

This is the way how the problem can be seen and how GRBL dies randomly. I do not see anything illegal in my sequence. And - of course - when I do something completely different with one of the GUIs (which can't check the pin) I do not see the problem, so how would that help in any way?

chamnit, 2018-08-20:
@Elmi77: Thanks for compiling with the Arduino IDE. It helps provide a solid basis.

Please keep in mind that compiling via Make is way more complicated than the Arduino IDE method for beginners who usually don't have any experience with command line. It's really not that hard. Copy a folder, open the example file, and click upload. 

G0 is fine if you're not trying to write settings in real-time like you described in the beginning. 

So you said that you don't see a problem when you don't check the (enable?) pin to rapidly react to when your last motion completes to avoid "stupid" parsing. So it means that it runs as it's supposed to when you're not trying to use Grbl in a way that it wasn't built for?  

Once again, I can't help you if you don't give me something I can test.

chamnit, 2018-08-20:
@Elmi77 : Set this Grbl setting to $1=255. This should eliminate your problem.

Elmi77, 2018-08-20:
How should this eliminate my problem? This pin is used to check if a motion is still running, with $1=255 I no longer can detect this! So you also could recomment me to not to use GRBL to get rid of this problem :-1: 

Sorry, I give up - you seem to not read what I'm posting here and even repeating it again and again does not help. So feel free to close this bugreport like you have done it with my previous ones. I personally do not expect any help here.

chamnit, 2018-08-20:
@Elmi77 : Please be respectful. I know you're frustrated, but it's difficult to know exactly what you are doing and how. This is not a paid support line and we/I do this in my free time. On top of that, it's difficult to get a full understanding of things when things are communicated back and forth through emails and postings. I have to ask you to do certain things to make sure that we are eliminating variables. Try to refrain from judgement before I offer an explanation of what may be causing your problem. If you hadn't noticed, I've been trying to help you for the last two weeks.

After writing my second to last response,  I realized that you were looking at the enable pin to trigger the response. If you read the wiki on the $1, you would have seen that Grbl holds the position for the $1 time in milliseconds after a motion completes before the stepper drivers are disabled. This delay uses a standard AVR delay tool, which may have contributed to the garbling the comm, especially if you are trying to communicate with it during this delay. If you set $1=255, then it removes the delay, but also doesn't disable stepper enable pin.

In normal use cases, Grbl CNC users don't use the enable pin as a way to signal and send a new command. You can't expect to use Grbl in a way that it wasn't built and designed for and demand that for it to work. You clearly have a specific use case that isn't for a CNC mill/router/laser. 

Issues like the one you are having are extremely hard to find, regardless of who you ask. We've been going through a very long process of elimination, trying to find where the issue is coming from. I think we may have found it, but sorry that you are giving up.

closed 2018-08-20 by chamnit

chamnit, 2018-08-20:
@Elmi77 : To be fair, the first time you mentioned you were doing something like observing the stepper enable-pin, as a way to detect when a motion completes, was mentioned for the first time today. This is not normal and should have been mentioned much earlier, when asked if you were not doing anything out of the ordinary.

cri-s, 2018-08-20:
This could explains the buffer overflow,.
I have this inside nuts_bolts c and h files.
I use it when i need to wait some external events, where timing is not
so critical.
As example the M-Fin from M20-28 (haas type ,  linuxcnc use more
complicated M63/M63/M66 for that)
Add this to the code and exchange delay_ms inside st_go_idle() with dwell  .
If the error now disappears, it was a overflow error in combination
with the '?' request
you have send. If not it's annother thing.

void dwell(uint32_t  i) {
     if(i>3600000) i=3600000; i*=1000;
     while (i>= 8000) {
     protocol_execute_realtime();
     if (sys.abort) { return; }
      delay_us(7975);
      i-=8000;
   }
     delay_us(i);
}

Chris

Elmi77, 2018-08-21:
@cri-s thanks for your feedback. The ?-command is sent only very rarely and I already check for the remaining buffer size. Nevertheless I tried your code, it seems to be useful for me in every case - at least to react on abort-commands even while waiting for the enable-output.

Unfortunately it did not solve the crash in GRBL (but I will keep it for above mentioned reason).

chamnit, 2018-08-21:
@Elmi77 : Please confirm that when you connect to the stepper enable pin to use as an input for your UI, that Grbl starts crash. Without it connected and using a standard UI, Grbl works fine. You mentioned this very briefly a few postings back.

109JB, 2018-08-21:
If you want to know when a movement command has completed, there are several way to accomplish this without monitoring the enable pin.  A very reliable way is to send a G4P0 command right after the command you want to know completed.  Then just wait for the grbl "ok" response to the G4P0 line.