gnea/grbl #508: Any ideas why GRBL resets $32 from 1 to 0 all by it'self.

StuartB4, 2018-08-19:
Apparently I posted this in the wrong place earlier, don't know why though.

This is strange.
When using LaserGRBL I go to GRBL Configuration and set $32 to 1 to enable Laser mode in GRBL 1.1f.
After clicking the Write button GRBL saves the setting.
I close LaserGRBL and reopen it and the setting is the same.
However, as soon as I connect to the correct Com port for my machine and then check the setting it has reverted back to the default $32=0.
Here's a video of what happens.
https://www.youtube.com/watch?v=i5VK8_LPmKA&feature=youtu.be

Any help on this would be really appreciated.

StuartB4, 2018-08-20:
Sorted now. It was a reset in LaserGRBL settings.

closed 2018-08-20 by chamnit

bdolph, 2018-12-28:
I have a problem. I baught grbl ardiuno shield 3.0 . I do now if i put my laser on whire on z + it activate my laser co2 tube. But what pin do i use to control power pwm

StuartB4, 2018-12-28:
Is the CNC shield a Protoneer?
If it is, then the Z- or Z+ is the control pin for PWM.
It is connected to Pin11 on the Arduino when the shield is plugged in, that is the PWM Pin on a version 3 Shield.
Both pins Z- and Z+ are connected together so it does'nt matter which one you use.