gnea/grbl #509: Latest development repository?

Elrond73, 2018-08-23:
Where can I find the latest GRBL-repository where current development is done? Or is GRBL no longer supported?

chamnit, 2018-08-23:
Here. The ARM version is being developed offline and will be available soon.

closed 2018-08-23 by chamnit

Elrond73, 2018-08-23:
GRBL is GPL, so please be so kind and publish the sources as required by the license.

chamnit, 2018-08-23:
@Elrond73 : As I stated, it'll be available soon. Getting rid of the GPL license has been one of the objectives as well.

Elrond73, 2018-08-23:
Sorry but you can't simply "get rid" of the GPL! Except you have written everything from scratch without reusing one single line of the original GRBL-code, it is still GPL and you have to publish it. So once again: publish the sources according to the GPL!

Schildkroet, 2018-08-23:
chill man
chamnit said, he will release it soon
What will be the new license?

Elrond73, 2018-08-23:
@Schildkroet it does not matter what he says, the GPL is quite clear in this point and one can't easily break it or think he is not bound to it - this is simply theft of public intellectual property!

Schildkroet, 2018-08-23:
@Elrond73 
There's no public release yet of grbl HAL and according to the GPL you don't have to distribute it

> In purely private (or internal) use—with no sales and no distribution—the software code may be modified and parts reused without requiring the source code to be released.

davidelang, 2018-08-23:
On Thu, 23 Aug 2018, Elrond73 wrote:

> Sorry but you can't simply "get rid" of the GPL! Except you have written 
> everything from scratch without reusing one single line of the original 
> GRBL-code, it is still GPL and you have to publish it. So once again: publish 
> the sources according to the GPL!

they don't have to give you code that they haven't given you a binary for, so 
stop pushing for the ARM code.

the regular code is here in this repo, so just clone it.

David Lang

109JB, 2018-08-24:
**Is the developer of a GPL-covered program bound by the GPL? Could the developer's actions ever be a violation of the GPL?** 

Strictly speaking, the GPL is a license from the developer for others to use, distribute and change the program. The developer itself is not bound by it, so no matter what the developer does, this is not a “violation” of the GPL.

davidelang, 2018-08-24:
On Thu, 23 Aug 2018, 109JB wrote:

> **Is the developer of a GPL-covered program bound by the GPL? Could the developer's actions ever be a violation of the GPL?**
>
> Strictly speaking, the GPL is a license from the developer for others [...]

as long as there is only one "developer", but as soon as others contribute to 
the code base, nobody has the privileged position to distribute anything under a 
different license.

But until distributintake place, GPL doesn't come into play. Someone can take 
GPL code and do whatever they want with it, but until they distribute it, nobody 
has to be given the source

David Lang