gnea/grbl #510: Disable EEPROM functionality?

oxygenic, 2018-08-27:
It seems for me, GRBL become instable whenever some EEPROM-accesses / setting of $-parameters is involved - also in case of strict handshake, means when only one $-command is sent and then the application waits until the related response arrives.

For my application it is no problem to send all $-parameters on start-up, so my question/feature request: is there a (compile-time) option to disable all EEPROM-stuff and to keep these parameters in RAM only?

Or any hints where to implement something like this?

cri-s, 2018-08-27:
To implement it for config values, not coordinate parameters, for that
there is a separate
patch.
This modification works as follow:
If you want ram only configuration change, then add 1000 to the
configuration number
Example
$1=255 ( this store it to eeprom)
$1001=255 (this is ram only modification)

You should have a method to load eeprom stored values, either add
read_global_settings() to mc_reset()  or add system command for that .
Adding it to mc_reset() resets the changes after M2 or any '~' sending.

Inside settings.c on function
settings_store_global_settings
change parameter from uint8_t  to uint8_16

Then add this line as first line.
  uint8_t store=1; if(parameter>=1000) { parameter-=1000; store=0; }

and at the end of this function  prepend
if(store)
to write_global_settings();

Update the settings.h file for the function prototype.
Inside system.c change
       return(settings_store_global_setting((uint8_t)parameter, value));
to
       return(settings_store_global_setting((uint16_t)parameter, value));

BR
Chris

oxygenic, 2018-08-27:
Thanks for your quick response! I want to keep all these parameters apart from EEPROM, which also can be listed by command $$.

> and at the end of this function  prepend
> if(store)
> to write_global_settings();

I think for my special case it is enough to remove the call to write_global_settings() since my application can ensure the parameters are always set on beginning.

chamnit, 2018-08-27:
@oxygenic : Grbl does not have $1000+ settings. This is a custom mod. Please see [this wiki excerpt](
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#eeprom-issues) on how to deal with EEPROM writes. I also suggest that you don't constantly write the settings. It'll fry your EEPROM much faster.

closed 2018-08-27 by chamnit

mayhem2408, 2018-08-27:
@oxygenic Just out of curiosity, what do you need to constantly change the $ settings. Are you using the same controller on different machines?

cri-s, 2018-08-27:
I needed to change motor enable or auto off and acceleration setting where
f value remains the same. The application was spray painting.

oxygenic, 2018-08-28:
@chamnit nobody said that RGBL has $1000+ settings!? Anyway, this issuse was already solved by @cri-s and hopefully this is also a workaround for the GRBL stability issues.

@mayhem2408 I try this because it seems the EEPROM-accesses are buggy and may cause issues with the serial communication - also when they are done in handshake as described in wiki. Unfortunately this issue happens very rarely so that I only can try different things and then wait for a long time. And it is up to the user to (re)configure these settings, so I can not be sure he never modifies them.

mayhem2408, 2018-08-28:
@oxygenic have you tried putting a G4 dwell after each EEPROM write. This might slow down the communication.
@chamnit 100% agree. This is not a "Stability Issue". And most people set their acceleration to the highest value there machine can handle and leave it there. This is the way all CNC controllers work. I have not seen a controller yet that alters acceleration on the fly. Not even commercial controllers. I assume this would also cause the planner all sorts of headaches.

oxygenic, 2018-08-29:
@mayhem2408 sorry, no I did not know of this possibility - I will try it next when the EEPROM-patch does not work.

And it for sure has nothing to with too high acceleration values or something like that, I'm working close to the default values and I do not change these values during a G-code sequence.

I'm even not sure if my problems really are related to the $-commands, as said I see these communication issues very rarely and just try to exclude possible reasons for this.

X3msnake, 2018-08-29:
are u using arduino clones?

there are known issues with the ch340 serial chip that most of these clones
use it.

dunno if this could be one  of those issues or if you had already looked
into it.

mayhem2408, 2018-08-29:
@oxygenic Plain and simple $ commands should not go in the Gcode file. This is not how it was designed. Are you possibly running a high per mm and high feed rate. That can cause instabilities as GRBL maxes out at around 30000 steps per second. Can you post for $ settings?

oxygenic, 2018-08-30:
@mayhem2408 please refer to my initial bugreport: I do not send $-commands in between, when I use them, I do a handshake strictly as prescribed by the manual. And I also use an original Arduino with the original USB chip.

Latest effect I see is a response like this where two state-messages overlap each other:

```
<Run|MPos:55.000,57.000,<Run|MPos:55.000,57.000,59.000|Bf:15,128>
59.000|Bf:15,128>
```

mayhem2408, 2018-08-30:
@oxygenic The reason I asked for your $ settings was to check that you are running within the limits for GRBL. If you are not updating $ values on the fly and having stability problems, then EEPROM is not your stability problem. If you run very high steps per mm and high feed rates will cause stability problems. I have experienced EEPROM corruption in some circumstances. I had a laser setup that ran 160 steps per mm, and max feed of 30,000mm/min. This is to high for GRBL to handle. Turns out I was maxing GRBL out at about 11,000mm/min. In relation to your initial post, I have not seen GRBL become unstable from updating $ values when done the correct way.

109JB, 2018-08-30:
The issues with the ch340 are well known, but the 16U2 that came on many original arduinos can also have issues. The difference is that the 16u2 can be fixed whereas the ch340 can't. 

https://github.com/grbl/grbl/wiki/Known-Bugs#usb-to-serial-transmission-errors

oxygenic, 2018-08-31:
@109JB I do not use the CH340, only original Arduinos. And problems become even more worse when I drop the USB chip and communicate via serial interface directly.

oxygenic, 2018-08-31:
@mayhem2408  Can "wrong" settings really cause problems with the serial communication? That's what I would call a bug.

These are my current axis-related settings:

$0=10
$1=75
$100=300.000
$101=300.000
$102=300.000
$110=600.000
$111=600.000
$112=600.000
$120=10.000
$121=10.000
$122=10.000
$130=5000.000
$131=5000.000
$132=5000.000

Limits are at 5000 (halve of what is specified to be GRBLs maximum) because the controlling application ensures the real limits are not exceeded. All other values are close to their defaults. Perhaps I should mention that I do not use the planner, I send one motion command and wait until it is completed, then I send the next command and so on.

Schildkroet, 2018-08-31:
https://github.com/gnea/grbl/issues/510#issuecomment-417228619

There is almost no way, that a status report overlaps, since it is printed inside 1 function (except the tx queue overflows).

So it's probably your application that has multi threading issues etc. and mixes up the received messages.

Use a proven application to test the behaviour. If it still occurs, post your program where the error occurs, so it can be reproduced.

oxygenic, 2018-08-31:
@Schildkroet it is a single-threading application on an other MCU which for sure has no threading problems and where the receive-function is called from exactly one position which is not from within an ISR. Anyway, thanks for giving feedback but my original question was already answered.

mayhem2408, 2018-08-31:
@oxygenic It's not so much wrong settings being a bug. It settings that are outside the limits on GRBL. For Example, your steps per mm of 300 means the max feed you can safely run is 6000. Your max feed settings are well below that, so that's not the problem.

cri-s, 2018-08-31:
> You can use this function that don't block interrupts and as result
> serial characters
> are not dropped.
>
> #include <avr/io.h>
>
> void EEPROM_write(unsigned int uiAddress, unsigned char ucData) {
>    for(;;) { unsigned char ocData;
>     if((ocData=EEPROM_read(uiAddress))==ucData) break;
> 	EEAR = uiAddress;
> 	EEDR = ucData;
> 	if(ucData==0xff)  EECR|= 1 << EEPM1; else
> 	if(!((ocData^ucData)&~ocData)) EECR|= _BV(EEPM1) + _BV(EEPM0);
> 	EECR |= 1 << EEMPE;
> 	EECR |= 1 << EEPE;
>        if (sys_rt_exec_state & EXEC_RESET) break;
>    }
> }
>
>
> unsigned char EEPROM_read(unsigned int uiAddress) {
>     while (EECR & (1 << EEPE));
>     EEAR = uiAddress;
>     EECR |= (1 << EERE);
>     return EEDR;
> }

And for the eeprom.h

void EEPROM_write(unsigned int uiAddress, unsigned char ucData) ;
unsigned char EEPROM_read(unsigned int uiAddress) ;
#define eeprom_get_char EEPROM_read
#define eeprom_put_char EEPROM_write

BR
Chris

2018-08-31 14:25 GMT+02:00, Cri S <phone.cri@gmail.com>:

chamnit, 2018-08-31:
@cri-s : The EEPROM write function in Grbl was taken directly from Atmel’s software framework from what I recall. I don’t see much reason to change it just for non-standard use cases, when they might create more risk in causing other EEPROM issues by turning them off.

cri-s, 2018-08-31:
The issue with the standard grbl eeprom is that it it need to write
114 byte of settings
for the case of N_AXIS is 3 and grbl 0.9 . Assume that a float value
and checksum need
to be written. This are 5 times over 3400 uS that interrupts are blocked.
3.4ms is the minimum time, typical time is stated as 3.6ms and worst
case is 5ms.
During  3,4 ms  35 transmitted serial bytes are lost on 115200 speed.

This code basically comes from 328 datasheet with some addition that
was omitted for
simplicity, but explained on datasheet.,

If you want correct you'r code, remove current cli/sei and place it as follow:
The reason for cli/sei is that  EEPE must be set inside 2 instruction clocks
  after setting EEMPE.  It is only attributable to programmer laziness
or if it don't matter
for application that cli/sei is set globally inside the function call.
This includes waiting
for write completion that in this case is the evil code that inhibit
receiving of serial characters. Instead on 9600 bps it is just 2
characters as worst case that are dropped.

        if( diff_mask & new_value ) {
                // Now we know that _some_ bits need to be erased to '1'.

                // Check if any bits in the new value are '0'.
                if( new_value != 0xff ) {
                        // Now we know that some bits need to be
programmed to '0' also.
                    cli();
                        EEDR = new_value; // Set EEPROM data register.
                        EECR = (1<<EEMPE) | // Set Master Write Enable bit...
                               (0<<EEPM1) | (0<<EEPM0); // ...and
Erase+Write mode.
                        EECR |= (1<<EEPE);  // Start Erase+Write operation.
                    sei();
                } else {
                        // Now we know that all bits should be erased.
                         cli();
                        EECR = (1<<EEMPE) | // Set Master Write Enable bit...
                               (1<<EEPM0);  // ...and Erase-only mode.
                        EECR |= (1<<EEPE);  // Start Erase-only operation.
                        sei();
                }
        } else {
                // Now we know that _no_ bits need to be erased to '1'.

                // Check if any bits are changed from '1' in the old value.
                if( diff_mask ) {
                        // Now we know that _some_ bits need to the
programmed to '0'.
                       cli();
                        EEDR = new_value;   // Set EEPROM data register.
                        EECR = (1<<EEMPE) | // Set Master Write Enable bit...
                               (1<<EEPM1);  // ...and Write-only mode.
                        EECR |= (1<<EEPE);  // Start Write-only operation.
                      sei();
                }
        }

BR.
Chris

oxygenic, 2018-08-31:
@chamnit how can sending of G-codes as described in the Wiki be a "non-standard usecase"?

@cri-s do you have an own GRBL repository somewhere? Seems it really worths to have a look into it!

chamnit, 2018-08-31:
@cri-s : I just want to make something explicitly clear to everyone. Grbl development here for the 328p in particular has pretty much stopped. Only critical bug fixes will be applied, primarily due to the fact that the 328p is completely maxed out. The remaining bytes of flash is reserved for only this. With the upcoming ARM release and re-write, I really see no reason to waste time on fixing minor issues like this one. 

Yes, the EEPROM write does write all of the settings at once, and it does have to stop communication for a few milliseconds. This is a known issue and a solution is written in the wiki about it. You can work around it very easily.

@oxygenic : The non-standard use case is writing to the EEPROM constantly. This was expected to be done only once or sparsely when first setting up the machine. Once again, just work around the issue. Please realize that while altering the interface a little may make your problem easier, it also breaks a lot of existing UI interfaces. This is something I'd much rather avoid. Any interface changes are reserved for major releases to try to keep the workload of the UI developers down. Many of them have tens of thousands of users and it would be a major headache for them to be constantly altering things. 

That said, don't fret. If I didn't already state (but have in other threads), the settings write issue should be eliminated in the next major version. All settings will be altered in RAM only until there is a specific EEPROM/NVM write command.

oxygenic, 2018-09-01:
@chamnit I do not wirte to EEPROM constantly - where do you have read this?

Btw: I do not agree with you regarding the 328. There are a lot of other ATMegas available which are very similar and offer more resources in terms of RAM, flash and EEPROM. And GRBL can be adapted quite easy to use them, simply by changing some defines. It is already designed in a way where other ATMegas can be used quite easy.

chamnit, 2018-09-01:
You state that you do in the very first post. 

You can argue as much as you want but the 328p is full. Literally a few bytes of RAM and a few dozen bytes of flash in certain critical configurations. 

Yes there are other AVR chips with more ram and flash. But none of these offer more speed which the next version needs.

oxygenic, 2018-09-01:
I said I use these functions but I never said I use them frequently! And I also said, I use them strictly according to the manual with full handshake. Seems the truth is these functions are broken in general?

chamnit, 2018-09-01:
No I’m not saying there isn’t a problem. You’re not the first one in the last couple of weeks that stated something similar. I find it very unusual that it just suddenly started to come up. I suspect there is some external set of circumstances, like a recent change in the compiler tools on the Arduino IDE, causing this but don’t have enough information to know exactly what. Any specific details would help.

mayhem2408, 2018-09-01:
@oxygenic There already is a version for the Mega. It's called GRBL-Mega. Right here https://github.com/gnea/grbl-Mega. It does have larger buffers and a couple of extra feature, but in general it is not faster. I use the Mega version because of the use of 16bit timers on the laser PWM output and the larger buffer help produce smooth rastering when lasering.

109JB, 2018-09-01:
> @109JB I do not use the CH340, only original Arduinos. And problems become even more worse when I drop the USB chip and communicate via serial interface directly.

OK.  So you don't use a ch340, and you use original Arduinos, but as stated, the 16U2 that comes on **_GENUINE ARDUINO UNO's_** and others can also have overflow errors on the USB chip. On the 16U2 it can be fixed by flashing a modified USB firmware to the 16U2. I posted the link.  If you have not flashed new firmware to the 16U2 you could very well have a USB overflow problem. This is a verified problem and just saying you use genuine arduinos doesn't mean you addressed it. The partial returned lines concatenated with another is exactly the problem that comes up with the USB overflow errors.

oxygenic, 2018-09-03:
@chamnit I do not use any specific. GRBL is the latest version from GIT, build tools are the ones that come with Ubuntu 16.04 by default (ArduinoIDE and avr-gcc), flashing is done via avrdude (and it does not make a difference if Optiboot in involved or not) and I do nothing more than sending a G0/G1 command and then wait until it has been completed before sending next command. So planner is not used, buffer sizes are always checked.

mayhem2408, 2018-09-03:
@oxygenic How are you waiting for the command to be complete and have you specifically disabled the planner? If you just waiting for the OK, it will only stop sending OK once the planner is full.

oxygenic, 2018-09-03:
I have not explicitly disabled the planner, it is not used implicitly because it gets only one command at the same time. Waiting for the end of a motion command is done via "?" commands which are sent every 50..150 msec. When it returns "Idle", next command is sent. But I plan to replace this by a check of the stepper-enable-output in future.

oxygenic, 2018-09-03:
What is weird in using normal G0/G1 commands and in using "?"-command to get the current movement state and the current axis positions? Is any of these commands buggy and therefore not to be used? This is not yet my final application but for testing I use such a scenario with one command at time - and the application is about to read/display the axis position during movement, that's why "?" is used.

I also tried the latest, precompiled release of GRBL but this did not make any difference.

Ubuntu for sure is not the problem, they provide the normal AVR-GCC like you can download it separately too.

But once again: my original question is answered, I come back as soon as I have some news/other questions regarding this issue.

chamnit, 2018-09-03:
It’s your unknown use case that is weird. Especially with no one (now 3 users in one month here and on other sites) is saying what it is. Each time they were expecting Grbl to work for their custom application but were using Grbl in ways it was not specifically built for, ie Grbl was built to expect a gcode stream generated by a CAM tool and do path planning. 

Try to remember that this is a open source project, not a paid product. You will run into problems when you step outside normal operation. It’s up to the generosity of the people who run the project to decide to donate their time to help you. So please try to be civil.

Also FYI, the Arduino IDE has used a very old version of AVR gcc tools in the past and have recently updated to a less old version. Ubuntu may come with a newer version. This is why I’m asking.

cri-s, 2018-09-03:
I have the same Problem When uploading modified Version and need to stream
the configuration down from serial interface. But I do it only very
occasional times.
If you want I can save the received characters inside ram and recover it
later in order you can see the differences of text.
With the eeprom diff patch this don't happen that resulting config is wrong.
Unfortunately its not easy upgrading remotely the firmware as the arduino
was modified for safety reasons that DTR don't make reset.
If machine with 12kg head travel at 0.5mt/sec or with more priced linear
guides 2mt/sec and the enable pin get disabled, at least machine bends it
structure after impact, calibrations are lost and machine must be
revisioned.
Controller switch from 16 microstepping to half stepping at 33.5khz and at
2mt/sec it stream at 35.2khz. Further during accel and decel the motor
voltage is increased.

mayhem2408, 2018-09-03:
@cri-s 33.5kHz and 35.2kHz are both above the stable limit of GRBL which is 30kHz. And I know from experience that going about that stable limit does cause problems.

cri-s, 2018-09-03:
It is full tested and timed.
The interrupt for 6 axis need 17uS for normal step operation and 24uS
as worst case.
Add 3uS as worst case for other interrupts gives full stable 37khz
operations and 4.5% processing power to gcode
parser/planner/protocol/report when streaming at 37khz
As the application don't use G2/G3 i have not timed that commands

chamnit, 2018-09-04:
Yes, the Grbl 30kHz step rate limit has a bit of margin on it. Primarily to give a little wiggle room for feature development. Running at 37kHz doesn’t surprise me, especially when you are not running arcs. Arcs and laser raster jobs are the worst case scenario.

oxygenic, 2018-09-04:
> It’s your unknown use case that is weird.

Not sure what is weird. What you say is "normal" is a stream out of G-Code commands in style

Gx/Gx/Gx/Gx/.../Gx/wait for end of operation (ok, this is simplified but it should show what i mean)

I want to do exactly the same but at the moment I'm in the phase of testing and setting up my system, so I'm sending a more simple stream in style

Gx/wait for end of operation

Both are the same by its principle and I really can't understand why you name the first "normal" while the second (which is simply a subset of the first) name "weird". I send streams of G-Code data for a plain and stupid CNC application, but right now I'm not in a state where the complex mechanism would work so I'm doing something much simpler. Is there a minimum of G-commands to be sent in order to do not stumble upon GRBL bugs? Or how else to understand this?

> You will run into problems when you step outside normal operation.

I would be really happy when you could explain where I step outside normal operation. I have a plain CNC application which works with G-Code commands. This is what GRBL is intended for, right? The only thing what is specific at the moment, is the fact that I have only one G-command instead of a bunch of them. I think this is something which always can happen also with a "normal" CNC application - e.g. when user decides to move the tool to a specific position - then only one single G0-command would be issued.

Schildkroet, 2018-09-04:
What kind of G-Codes are you sending? Pls provide some examples there the problems occure.

oxygenic, 2018-09-05:
F, G0 and G1

oxygenic, 2018-09-13:
Short update: when I send something like

F3000
G1X10Y20Z30
G1X0Y0Z0

I sometimes can see some kind of memory leak (values returned by $$ contain random crap).

It does not seem to be caused by the relative high value of "F3000", the same happens with "F300" or "F30". More than this it seems to happen only after the last command. Means when I send

G1X10Y20Z30
G1X0Y0Z0

memory is corrupted after the "G1X0Y0Z0". When I send 

G1X10Y20Z30

and then wait for the end, it is corrupted at this point. So this fits to my previous operation mode where I send one command and then wait until motion is finished - and sometimes get in troubles afterwards. Not sure yet if a permanent sending of "?" has an influence on this...

oxygenic, 2018-09-24:
Short update: the behaviour mentioned above is random. Sometimes the $$-values are damaged after sending this sequence once, sometimes I can execute it 100 times without noticing any problems.