gnea/grbl #514: Error compile

sfinexer, 2018-09-03:
Cause of error in the following
config.h line 30:
#include "grbl.h" // For Arduino IDE compatibility.

But in #include "grbl.h" of a line 65+
// COMPILE-TIME ERROR CHECKING OF DEFINE VALUES:

Reason:
config.h announces variables to a poszha what the connected file #include "grbl.h" does check of definitions.

sfinexer, 2018-09-03:
I suggest to change sequence
config.h:

+#include "grbl.h" // For Arduino IDE compatibility.
#ifndef config_h
#define config_h
-#include "grbl.h" // For Arduino IDE compatibility.

jdt1, 2018-09-05:
I think I'm experiencing the same issue. Did you find a work-around? 

The complete error output for me is
```
Arduino: 1.8.6 (Windows 10), Board: "Arduino/Genuino Uno"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\<SNIPPED>\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10806 -build-path C:\Users\<SNIPPED>\AppData\Local\Temp\arduino_build_360569 -warnings=all -build-cache C:\Users\<SNIPPED>\AppData\Local\Temp\arduino_cache_968448 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\<SNIPPED>\grbl-master\grbl\examples\grblUpload\grblUpload.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\<SNIPPED>\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10806 -build-path C:\Users\<SNIPPED>\AppData\Local\Temp\arduino_build_360569 -warnings=all -build-cache C:\Users\<SNIPPED>\AppData\Local\Temp\arduino_cache_968448 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\<SNIPPED>\grbl-master\grbl\examples\grblUpload\grblUpload.ino
Using board 'uno' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Detecting libraries used...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10806 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "C:\\Users\\<SNIPPED>\\AppData\\Local\\Temp\\arduino_build_360569\\sketch\\grblUpload.ino.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10806 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "-IC:\\Users\<SNIPPED>\\Documents\\Arduino\\libraries\\grbl" "C:\\Users\\<SNIPPED>\\AppData\\Local\\Temp\\arduino_build_360569\\sketch\\grblUpload.ino.cpp" -o nul
In file included from C:\Users\<SNIPPED>\Documents\Arduino\libraries\grbl/config.h:30:0,

                 from C:\Users\<SNIPPED>\grbl-master\grbl\examples\grblUpload\grblUpload.ino:1:

C:\Users\<SNIPPED>\Documents\Arduino\libraries\grbl/grbl.h:68:4: error: #error "Required HOMING_CYCLE_0 not defined."

   #error "Required HOMING_CYCLE_0 not defined."

    ^

C:\Users\<SNIPPED>\Documents\Arduino\libraries\grbl/grbl.h:108:4: error: #error "WCO refresh must be greater than one."

   #error "WCO refresh must be greater than one."

    ^

C:\Users\<SNIPPED>\Documents\Arduino\libraries\grbl/grbl.h:111:4: error: #error "Override refresh must be greater than zero."

   #error "Override refresh must be greater than zero."

    ^

Using library grbl in folder: C:\Users\<SNIPPED>\Documents\Arduino\libraries\grbl (legacy)
exit status 1
Error compiling for board Arduino/Genuino Uno.
```

This is after removing Arduino and installing the latest version 1.8.6, using grbl 1.1.

sfinexer, 2018-09-06:
Of courseconfig.h line 27-31:Was:#ifndef config_h#define config_h#include "grbl.h" // For Arduino IDE compatibility. Became:#include "grbl.h" // For Arduino IDE compatibility.#ifndef config_h#define config_h It is possible and differently, just like that quicker.

-- Здравствуйте,

chamnit, 2018-09-06:
@sfinexer : Go into Preferences and disable the "Aggressively cache compiled core" option. This should get rid of the compiler warnings about things like "False" being redefined.

If you are getting errors like  `#error "WCO refresh must be greater than one."`, then you either have incorrectly imported Grbl or have an existing Grbl installation in your Arduino path. Make sure you select the grbl source directory inside the unzipped folder, not the unzipped folder itself.

sfinexer, 2018-09-06:
@chamnit I have checked right now. Doesn't help.
The same mistake at me.
But if to change the file as has stated above, compilation takes place without mistakes.

sfinexer, 2018-09-06:
@chamnit: I have stated above that a mistake not only at me!
You what you can't check??? Or me to make video?

chamnit, 2018-09-06:
@sfinexer : You're not using the GrblUpload example, as the instructions state. It only has one include for grbl.h. There is a specific reason for this. It hacks how the Arduino IDE works to allow you compile Grbl's custom code.

closed 2018-09-06 by chamnit

jdt1, 2018-09-07:
I had somehow mixed up the files. I was using the example code from one .zip package, but the library files from another. When I used all files from the same package, the issue was resolved. Just a silly mistake in my case.