gnea/grbl #516: Loosing Steps

algo1980, 2018-09-18:
Hi everyone, I am ending up in a very confusing situation. When I jog the machine while the spindle is turned off it works well but as soon as I kick in the spindle the machine starts to lose steps. For the electronics, they are placed in a metal enclosure which is grounded.

romychs, 2018-09-18:
I think the spindle eats up all the power of the power source, if it is common for him and stepper motors.  Lack current  to stepper motors to hold the position.
If you think this is interference, try soldering the ceramic capacitor 0.1uF-1uF between the motor terminals.

algo1980, 2018-09-19:
thank you romychs,
do you mean i should connect a ceramic .1-1uf cap on both the terminals which are A+ A- and B+B-?

swarfer, 2018-09-19:
please read this
https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches
and this 
http://www.mycncuk.com/threads/9504-Why-star-ground

separate wiring does not ensure the interference does not happen because the interference is transmitted through the air just like a radio signal.   if your spindle has brushes then it makes sparks, and sparks make a lot of radio noise (the first radio transmitters were just spark gaps!).   any wire that has varying electricity in it will transmit, and any other wire will pick up some of that transmitted energy.

romychs, 2018-09-20:
**Algo1980**, I had in mind a capacitor between of spindle motor leads. **Swafer** well described the cause of possible interference.
Re-read the question again, the steps are lost when you kick the spindle (by hands, for example), do not turn it on or off? Then, all I said above will not solve the problem. Try adjusting the hold current of stepper motors. Try reducing the half-step mode, from 1/16 to 1/8 for testing only.

zeevy, 2018-09-26:
This may depend on how you wired and routed the spindle wires. you should avoid routing spindle wires along with other signal wires

wmarkow, 2020-04-20:
>as soon as I kick in the spindle the machine starts to lose steps

When you manipulate spindle speed you need to wait a bit of time by adding some dwell commands. For me **G4 P0.5** did the job.
See also [here](https://github.com/grbl/grbl/issues/951).