gnea/grbl #517: Running grbl at 9600 baud

classbproject, 2018-09-20:
I know that the v1.1f version runs at 115200 baud but I seem to be running into Serial comms issues with this on bCNC. The program keeps failing mid-cycle with a Serial comms error. Folks over at bCNC suggested that this might be because of the CP2104 USB-UART bridge in my hardware. I find that unlikely since it works flawlessly with the OpenCNCPilot. However, to definitely rule out the CP2104 as the source of my issue, I was wondering if it is **_advisable_** to run the current version of grbl at 9600 baud. I remember reading somewhere a comment by @chamnit that you should not but I may be mistaken.

oxygenic, 2018-09-24:
In general: you can change the baudrate in config.h by modifying the value for BAUD_RATE. But I would guess you stumbled upon an other, unsolved issue I notice too. There are some closed issues regarding this and until now there seems to be no solution for these communication problems. At least for the problem I see here (too), I can confirm slowing down the communication speed does not change anything.

classbproject, 2018-09-24:
I can confirm that slowing down the comms rate actually created problems with GUIs that were working glitch-free before. I'm back to 115200 baud. Many on the bCNC page have suggested that the CP2104 might be buggy but I find that unlikely since they seem to be having these kind of issues with the CH340G and the FT232RL.

chamnit, 2018-09-24:
You should be able to reduce it down to 9600 without much issue. Older versions of Grbl ran at this speed and not much has changed with the serial code since then. 

With regard to comm issues, there hasn’t been a clear set of cases of what the problem is. The ones reported seem unrelated to each other or isolated. People have had trouble providing a reliable test that I can reproduce so that I can troubleshoot it. So problem solving has been limited to slow and inefficient issue posts, which can get frustrating very quickly. 

That said, I’ve played with the CP2104 without issues. There are a number of external sources that can effect connectivity and data transfer. For example, if the USB cable does not have a ferrite core to reduce EMI or is near sources of EMI, data transfer can fail between the USBserial and host computer. Usually running the job with nothing else powered except the Arduino only and preferably disconnected from everything will typically eliminate EMI as a factor. 

I’m still trying to compile what the reported comment issues so please provide your Grbl $$ settings, $i build info output, OS you are running, and detailed info on how you compiled Grbl and with what version of Arduino IDE. Thanks!

classbproject, 2018-09-24:
I ran the gcode without the end mill but with the motors on and the program stopped randomly. I then ran the gcode with the motors turned off and this time it randomly stopped at a different point in the code. I ran the same code on OpenCNCPilot and it completed without a hitch with everything turned on. I don't think EMI is an issue because my cables are shielded and the end stop signal has to go through a low pass filter followed by a Schmitt trigger. 

I too don't think the CP2104 is the cause but I opened this issue just to rule out the possibility since the folks at bCNC seem to think that is the source. 

I compiled grbl by the book so to speak. First added the `grbl` folder inside the `grbl-master.zip` to the `libraries` folder under the Arduino sketches folder. I then opened the `grblUpload.ino` file in the Arduino IDE, compiled it and then uploaded it to the 328P. Using IDE version `1.8.6`.

`$$` output is:

	$2=0
	$3=1
	$4=1
	$5=1
	$6=0
	$10=1
	$11=0.010
	$12=0.002
	$13=0
	$20=0
	$21=1
	$22=1
	$23=3
	$24=300.000
	$25=1425.000
	$26=250
	$27=2.000
	$30=12000
	$31=0
	$32=0
	$100=80.000
	$101=400.000
	$102=400.000
	$110=3000.000
	$111=2200.000
	$112=1000.000
	$120=200.000
	$121=200.000
	$122=100.000
	$130=725.000
	$131=190.000
	$132=15.000

`$4=1` because my Toshiba drivers require an active high enable signal.  

`$i` output is:
	Response:[VER:1.1f.20170801:]

chamnit, 2018-09-24:
@classbproject : Are you running windows? Also, the fact that OpenCNCPilot works and it runs on .net suggests that something might be wrong with bCNC's python serial library. I recall there has been some issues with the pyserial library in the past, but can't remember exactly what. I'm pretty sure you had to use a certain version for it to work properly. This was a long time ago. It's feasible that some new version of pyserial (and maybe OS dependent) might be the culprit.

chamnit, 2018-09-24:
@classbproject : Another way to check is to test with another GUI that uses some other serial library or written in a different language. UGS is written in Java, not sure what it uses for its serial interface, but it'd be worth a shot to see if the problem doesn't occur there.

classbproject, 2018-09-24:
Yes, running Windows 10. I installed whatever was the most recent version of pyserial was. Don't really know what it is. Also there seem to be two versions of bCNC - one by Harvie and one by original developer. I believe the Harvie version is a little ahead of the original version in that it does not need certain libraries for displaying data such as the auto leveller probing points, among other things. I tried with both and it failed with both. 

I will try with UGS and report back. Not sure what is used in UGS 1.0.9 but the UGS 2 nightlies give you the [option](https://github.com/winder/Universal-G-Code-Sender/issues/940) of JSSC and JSerialComm but the CP2104 only works with JSerialComm. With JSSC selected it does not detect the device.

chamnit, 2018-09-24:
@langwadt : It sure would. When things were running at 9600 baud, status report polling was about 1 per second, rather than 10 or so per second now. It would definitely need to be reduced.

oxygenic, 2018-09-25:
> People have had trouble providing a reliable test that I can reproduce so that I can troubleshoot it.

Please refer to my (closed) issue, there I gave you a sequence which at least leads to a memory leak damaging the $$-values and which probably also can corrupt the communication.

chamnit, 2018-09-25:
@oxygenic : Yes, there is a sequence, but no details on anything else, i.e. test setup, comm specifics, other external factors like wiring, a program that I can stream that will repeatably show case the issue, etc. I've requested this a number of times. But, if you'd like to talk about this more, we can do this in a new thread or your old one. Lets not hijack this one, because it seems unrelated to your issue.

oxygenic, 2018-09-26:
I can't give you what does not exist: the issue is NOT reproducible reliable, it is random. So probably a timing issue? In this case I of course can't do any magic to make it reproducible securely, a random bug stays random and is always difficult to debug.

Everything else is standard, the G-code-sequence can be found in the bug report, the device is a plain and original Arduino Nano without anything connected, the sources are either your latest GRBL-version out of GIT or my variant with the disabled EEPROM-access (which does not have an influence on this, it happens with both). Communication is done with 115200 or 57600 baud, this also does not make a difference. So you can go with a plain and simple default set-up and this G-code-sequence.

classbproject, 2018-09-26:
Ok so I tried with UGS 1.0.9 and it worked perfectly. In fact I ran 4 jobs and they all ran perfectly without any glitches at 115200 baud.

chamnit, 2018-09-26:
@classbproject : Ok. This seems to confirm something is up with bCNC. It could be some incompatibility with your computer (OS, Python version, external tasks), a serial driver issue, or a bug in bCNC. 

The bCNC folks may ask for more proof. There is another GUI that uses nodeJS drivers called CNCjs. If it works, this would provide definitive proof for them. However, the installation of Node isn't very straightforward or pretty. I'd hold off unless they ask for it or are comfortable with taking that extra step.

Otherwise, I'll consider this matter closed since we've answered the original question and seemed to have verified that its not Grbl. Let me know what happens regardless. I'm tracking serial comm issues pretty closely, since there has been random related issues cropping up recently.

closed 2018-09-26 by chamnit

Schildkroet, 2018-09-26:
@oxygenic 
I tested your sequence but didn't run in any problems.
Please get some approved gui like UGS etc. and test your grbl setup with it. If you can reproduce the issue, post your exact setup and steps, so the problem can be reproduced.
If you are not willing to do so, stop complaining about it in every thread and accept the current state.

oxygenic, 2018-09-27:
@Schildkroet 

> I tested your sequence but didn't run in any problems.

How often did you try? As mentioned sometimes it happens after a few loops, sometimes you need to try >100 times to see the issue. 

> Please get some approved gui like UGS etc. and test your grbl setup with it. If you can reproduce the
> issue, post your exact setup and steps, so the problem can be reproduced.

I do not use any GUI but send the sequence directly via serial console/script and wait until it is finished by checking the "Idle" information:
- the total length of the sequence is smaller than 128 bytes so it for sure does not cause a serial buffer overflow
- the total amount of commands is smaller than 14, so it for sure does not cause a command buffer overflow
So what difference could a GUI make exactly?

> If you are not willing to do so, stop complaining about it in every thread and accept the current state.

I do not complain "in every thread" but I comment obvious things - and when somebody claims there are no usable reports available regarding a specific issue, I think it is more than fair to let me give an answer when this is not true!