gnea/grbl #519: Looking for good jogging implementation for grbl.

hdo, 2018-09-26:
Has anyone seen a really good implementation of the 'new' grbl 1.1 jogging feature.
I really like GRBL for my machine but jogging with bCNC is really a mess and very jerky.
I tested the offline DDSCV CNC controller and their jogging feature with MPG is really smooth.
I have yet to see a similar implementation of such smooth jogging for GRBL. Can anyone recommend
a gcode sender (preferably open source) with good jogging implementation? 
There is a good explanation on the wiki page but my own experiments weren't promising.

zeevy, 2018-09-26:
Try this mobile application 
https://github.com/zeevy/grblcontroller

hdo, 2018-09-26:
Thanks for that. There is really demand for sending G-Code via Smartphone? However, i'll take a look at that.I don't have a bluetooth module, wondering that android have any kind of driver for USB serial. I would really like to see this in action ;-)

> Try this mobile application
> https://github.com/zeevy/grblcontroller

zeevy, 2018-09-26:
Any android phone with version 4.4 (kitkat) or above supports usb otg connections.
App will work in USB otg mode if Arduino has latest serial firmware installed.

hdo, 2018-09-26:
Are you using small steps for jogging or a very large one with jog cancellation?

zeevy, 2018-09-26:
Not using one large step with jog cancellation.
Step size and feed rate can be configured by user ( so you can use large step mode )
Pressing and holding the button will send continues jog commands to the machine until planner buffer is filled. Once the planner buffer is filled it will wait for more room before sending another jog command.

hdo, 2018-09-26:
I see thanks for clarification. Last question though without deeping to much into your code. How do you check whether buffer is full?

zeevy, 2018-09-26:
Reading the status report from grbl
$10=2

hdo, 2018-09-26:
> Reading the status report from grbl
> $10=2

Thanks ;-)

Sunnysky, 2018-09-27:
GRBLpanel.exe is the best for Jogging and Macros and buttons , keys and loading files to execute.  USB 115kb or IP address interface

hdo, 2018-09-27:
Thanks for the hint. I will take a look into that. It runs on windows so no use in my productive environment though.

> GRBLpanel.exe is the best for Jogging and Macros and buttons , keys and loading files to execute. USB 115kb or IP address interface

Sunnysky, 2018-09-27:
You can run it any cheap old Windows PC and probably WINE,  and then remote control it from your mobile on Teamviewer if you wanted.   If want a productive GUI , it doesn't get any better

hdo, 2018-09-27:
I'm working on a 'mimimal' fool proofed read-only linux system for my set up (similar to DDCSV CNC offlien controller). In the past i found windows very unreliable for such tasks ;-). My Linux system an a raspberry pi is cold booting up in under 15 seconds which i can't get with an old windows system.

Sunnysky, 2018-09-27:
I could get 15 sec  on my old XP machines with 1GB of RAM, but you have to know how to trim the fat  (not FAT), beside your criteria was excellent JOG features and I told you the best GUI. period.  This is not a DSL contest.

Boot time , who cares, But the less overhead and features XP can boot in 10 seconds with the setup. Just use Autoruns.exe and a few other tools to configure.

for this on a UNO with a CNC shield for and a 1x1m gantry cost me<$500 with a 1W laser

hdo, 2018-09-28:
However, i really like the UI design of GRBLpanel. bCNC looks kinda old but that's what you get with TKinter.

jon-daemon, 2020-04-13:
I use UGS on Linux Mint and I have setup shortcuts for jogging. I remapped the keys from a usb gamepad with the same shortcuts and I move my machine with the gamepad buttons. I would like to move my machine with the analog stick of my gamepad and stop moving when I stop touching the joystick . Is there any implementation for that? Is there any script or app?

Sunnysky, 2020-04-13:
@jon-daemon I don't know if @hdo still has this problem but I understand it very well with my electronics servo and control theory background.  All electro-mechanical systems have resonant frequencies which greatly affect this issue due to mass and slack in the drive train.

   The bandwidth of energy of the stepper acceleration may be easily in the GRBL.PANEL code table for maximum g, maximum v regardless of the jog size x.  The rate of change of g is also called a "jerk" which is the essence of this problem.  Any slack in the belt or drive train causes hysteresis which due to the slack may cause a sudden jerk. Besides this, in the GRBL code any great software implementation, such as GRBL PANEL has this constant which is non-volatile and may be easily modified in a table then is uploaded to the Arduino g-code interpreter to execute with any command such as a jog.  One can increase the maximum velocity, v_max of a large jog while reducing the maximum acceleration g_max by "trial and error" to optimize or compute it.

One way is to record the sound and FFT using Audacity (free SW) on a PC, to determine the lowest resonant frequency and then use the standard formula of rise-time **Tr=0.35/f (-3dB)**. This rise time is the time to make a short step or jog which is controlled by the acceleration, g of physical steps to arrive back and forth or half of the inverse repetitive jog back and forth frequency 1/2 * 1/f = T  controlled by g in your software.  Thus you can reduce the excitation energy accelerating and braking to these jogs to get a smooth response by attenuating the spectrum amplified at your resonant frequency.

I hope this makes sense. 

It is why I love GRBL PANEL so much as this is so easy to change and allowed me to achieve a maximum velocity of 2 meters per second by reducing g_max or a (acceleration) and increasing velocity constant v_max for any seek step size. 

On the earliest Hitachi/NPL 5.25" hard disk drives with steppers, they used this effect and also added a thin flat circular plastic disk about the same diameter as the stepper motor to act as a shock absorber. It had a thin brass foil disk surrounded with oil to act as a low mass rotary damper. It worked amazing well. Unfortunately, none of the floppy drives I have ever seen used this g and v control servo method, except perhaps Wozniak's Apple ][ to seek fast with low jerks, but still much slower than this Hitachi/NLP HDD. This has nothing to do with the buffered step rate pulses which can be 100kHz.

Sunnysky, 2020-04-13:
@jon-daemon I tried UGS but never liked it.  on GRBL PANEL you can use the 4 arrow keys to jog and with your keyboard repeat lag and rate use that as a joystick.   I imagine one can use the analog joysticks position to even control velocity with a dead-band calibrated at the neutral position.  But I don't know your interface or the GCS firmware to determine how, but it should be possible to control servo speed and direction.

winder, 2020-04-13:
@jon-daemon a joystick plugin was added to UGS recently, if you have a chance to try it let us know how it works for you: https://github.com/winder/Universal-G-Code-Sender/pull/1385

SailWithChips, 2020-04-16:
Thanks a lot for your message and effort @winder . Just make some first tests and seemed to works nicelly. Tomorrow will test deeply. Your work is much appreciatted ;)

jon-daemon, 2020-04-17:
I also checked it for a few minutes and works fine. Thank you for your work. The only thing I noticed is it doesn't recognise correctly the axis on my gamepad joysticks. It sees Left joystick Y as Right joystick X and vise versa. It's not an issue since I can match any command I want for each axis.

fbonin5491, 2023-09-01:
i don't like Autolevel in UGS-Platform , and UGS does not have any.