gnea/grbl #520: limit pin invert

sfinexer, 2018-09-27:
When I put $5 = 1 limit switch vanish.

sfinexer, 2018-09-27:
I have a beginning of axes in the left bottom corner.
I put 23$ =7 I do search, but then I can't work because the position x/y more than 0 at $20 of =true doesn't work.

sfinexer, 2018-09-27:
that is it turns out what the machine position will always be from - *** to 0?

sfinexer, 2018-09-28:
I don't align Z axis.
//#define HOMING_CYCLE_0 (1<<Z_AXIS)                // REQUIRED: First move Z to clear workspace.
#define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))  // OPTIONAL: Then move X,Y at the same time.
But I receive an error of a soft limit of axis Z.

sfinexer, 2018-09-28:
To receive positive and negative coordinates on axis Z to me неоходимо to disconnect soft limits on this axis.

109JB, 2018-09-28:
Just use work coordinates and you can have positive coordinates.  Look up G54-G59 in any G-code reference.

sfinexer, 2018-09-28:
@109JB Can make an opportunity to disconnect soft limits on separate axes?