gnea/grbl #522: Missed steps at movement start

toktuff, 2018-09-30:
I would like to check if there is some way to work around this issue I'm having.

I'm running my CNC with external drivers (DM542T from stepper online to be exact). At the very start of the motion (or with short motions, like 1mm jogs) it loses some steps.

After some testing I figured out that if I let the steppers be powered up all the time ($1=255) then this wasn't an issue.

It would seem to me that these drivers would like to have a delay between being enabled and the first step signals.

Is this something that is possible to configure?

lnelson214, 2018-10-07:
I don't think there is an initial delay setting, but is there some problem with leaving them enabled all the time?

toktuff, 2018-10-08:
The problem is that they will be getting high current from the controllers and generating heat. It feels unnecessary the have the steppers hot when idle.

My Z axis is a bit extra problematic since it needs full current all the time. (The other axes can use a "half current when idle" feature available on the drivers).

toktuff, 2018-10-13:
I'll keep my current configuration.

closed 2018-10-13 by toktuff